ARMmbed/mbed-os-example-mesh-minimal #23: Fresh checkout results in undefined references

stevew817, 2016-10-03:
I did a fresh checkout of this demo app, and did just what was written in the readme. However, when I try to compile, I get undefined references at link time to:

- arm_net_phy_register
- arm_net_phy_unregister
- net_init_core
- a bunch of arm_nwk_* and arm_net_* functions
- socket_* (open, close, read, sendto, connect, bind)

I'm guessing somehow Nanostack itself is not getting linked. Did I miss something?

c1728p9, 2016-10-04:
Hi @stevew817, what target are you compiling for? There is a known issue preventing nanostack from getting linked into cortex-m3 devices. A fix for this can be found here: https://github.com/ARMmbed/mbed-os/pull/2901

stevew817, 2016-10-04:
Trying to compile for either K64F (M4) or EFM32GG (M3). The fix doesn't solve the issue, but moving the sal-stack-nanostack folder from features/net/FEATURE_NANOSTACK to the root of the app does.

Which is strange, since the build system does pick up everything else in the FEATURE_NANOSTACK directory, just not sal-stack-nanostack.

SeppoTakalo, 2016-10-04:
I cannot replicate this problem on K64F.
I also tried `EFM32GG` but my toolchain tells it is unknown platform.

That M3 build is now a known problem.

c1728p9, 2016-10-04:
@stevew817 what version of mbed-os and this example are you using? Can you attach the sha of each?

stevew817, 2016-10-05:
sure, mbed os is at bf8d821, example at efc3260.

SeppoTakalo, 2016-10-05:
On a git commit `efc3260` and mbed-os on git commit `bf8d8211` I can still compile for K64F using GCC_ARM.
Are you sure that this was a clean checkout?

Those function names directly points to Nanostack and related RF drivers and I know that there has been refactoring done on the RF interface between mbed OS and Nanostack. Most clearly all those `rf_*` prefixed functions have been moved to `static` namespace.
So this issue was probably visible last week (or week before that) when mbed OS is updated but Atmel RF driver is still lacking behind.

For us, this is a clear indicator that we should prepare our releases on the `-rc` branches instead of `master` because there might be developers importing the application just in the middle of the refactoring work. So habit of using `master` as a working branch is not acceptable.

Thanks for reporting this. I will close this now, as I cannot replicate it. Feedback received.

Br.

closed 2016-10-05 by SeppoTakalo