ARMmbed/mbed-os-example-mesh-minimal #26: Can't export the project with online compiler

balamk07, 2016-10-05:
Hi,

I couldn't export this project to any of the external toolchains supported by mbed online compiler. Export option works with mbed-cli, but the project file generated by it doesn't compile for KDS or IAR toolchain. mbed cli pulls in all the source files under mbed-os as well as the protocol stack directories causing many compilation errors. Could you please fix the export option for this project so that it can be exported to different toolchain from mbed online compiler?

Thanks,
Bala

SeppoTakalo, 2016-10-25:
This example seems to export with the offline tool.

@theotherjimmy Can you help, what is the problem here?

I tried to import this app to online tool. Works fine, builds fine.

Then I tried to export:
[image]

hasnainvirk, 2016-12-28:
@balamk07 Is it still valid ?

balamk07, 2017-01-03:
@hasnainvirk I just tried to export the project with mbed online compiler. Unfortunately the issue still exists

hasnainvirk, 2017-01-03:
@screamerbg Can you please help here ?

SeppoTakalo, 2017-06-14:
Closing this as this is a online compiler issue, not related to this app.

closed 2017-06-14 by SeppoTakalo