ARMmbed/mbed-os-example-mesh-minimal #34: Connection failed

muddog, 2016-12-09:
Hi

I just setup two FRDM-K64+MCR20A boards, built image with 6LowPAN MESH configured to run this demo. Looks like it requires the Boarder router or Gateway to connect. From the log "No Beacons" should mean we do not have a PAN Coordinator right? As this example name: meah, it should work without a border router or gateway. Am I right?

The log attached:
```
Connecting...
[DBG ][core]: Allocate Root Tasklet
[DBG ][6lo ]: P.Init

[DBG ][evlp]: event_loop_thread
[DBG ][core]: NS Root task Init
[DBG ][sck ]: Socket Tasklet Generated
[DBG ][MCR20A]: rnd:59, MAC 64-bit: 00:05:a5:60:38:a3:f0:09
[DBG ][m6La]: init()
[DBG ][m6La]: connect()
[DBG ][sck ]: Socket id 0 allocated
[DBG ][m6LND]: Link-layer security NOT enabled.
[DBG ][m6LND]: Channel: 12
[DBG ][m6LND]: Channel page: 0
[DBG ][m6LND]: Channel mask: 4096
[INFO][addr]: Address added to IF 0: fe80::205:a560:38a3:f009
[DBG ][addr]: LL64 Register OK!
[DBG ][rout]: Added route:
[DBG ][rout]:                    fe80::/64  if:0 src:'Static' id:0 lifetime:infinite
[DBG ][rout]:      On-link (met 128)
[INFO][m6LND]: Start 6LoWPAN ND Bootstrap
[DBG ][mle ]: MLE Tasklet Generated
[DBG ][6Bo ]: Start Active Scan
[DBG ][mlme]: chan page 0, mask 1000
[DBG ][mlme]: MAC: Start MLME scan
[DBG ][mlme]: Scan channel 12
[DBG ][mlme]: BEA REQ tx
[DBG ][mlme]: Scan Complete..Halt MAC
[DBG ][mlme]: Trig confirm
[DBG ][mlme]: Active Scan Result
[DBG ][6Bo ]: Mac scan confirm:No Beacons
[DBG ][mlme]: 00
[DBG ][mlme]: Free Response
[DBG ][6Bo ]: Network Bootsrap Start Fail
[DBG ][mleS]: Free Socket
[DBG ][mlme]: RESET MAC PIB
[DBG ][m6LND]: app_parse_network_event() 3
[DBG ][m6LND]: Link Layer Scan Fail: No Beacons
```

hasnainvirk, 2016-12-09:
Hi @muddog ,

6LoWPAN network is not fundamentally a peer-to-peer network. The fundamental assumption here is device-to-server. Also, 1EEE 802.15.4  spec suggests that you should have at least one PAN-coordinator (gateway/border router) in your mesh network. 
However, on the other hand, a Thread device can start a standalone network on its own by paying sufficient costs in terms of complexity, i.e., intelligently deciding when to form its own network, network discovery, joining another network later on, migrating children to new network etc etc.

Shrugging off the complexity and adhering to simpler spec definition, a 6LoWPAN mesh device (router/host) seeks for a coordinator while bootstrapping. Essentially, every device starts its bootstrap as a host and after finding a beacon from a coordinator uopgrades itself to a router (if routing capability turned on). 
I hope this will clarify. If it does please close the issue, as it is not really related to the application at hand.

muddog, 2016-12-10:
Thanks for clarify!

closed 2016-12-10 by muddog