ARMmbed/mbed-os-example-mesh-minimal #37: Connection failed when using NUCLEO_F429ZI with ATMEL RF as Thread mode

MarceloSalazar, 2016-12-09:
Tested 5.3-RC3 branch with NUCLEO_F429ZI platform.
Thread BR on mbed GW (aka firefly) - tested both Thread dynamic and static version from here:
https://github.com/ARMmbed/mbed-os-example-client/tree/mbed-os-5.3-rc3/GW_Binary

`mbed_app.json`:
```
       "mesh-type":{
            "help": "options are MESH_LOWPAN, MESH_THREAD",
            "value": "MESH_THREAD"
        }
    },
    "target_overrides": {
        "*": {
            "target.features_add": ["NANOSTACK", "THREAD_ROUTER", "COMMON_PAL"],
            "mbed-mesh-api.thread-device-type": "MESH_DEVICE_TYPE_THREAD_ROUTER",
```

Serial output with trace enabled (also tried with trace disabled)
```
Connecting...
[DBG ][nsth]: connect()
[DBG ][evlp]: event_loop_thread
[DBG ][m6Thread]: PANID defa
[DBG ][m6Thread]: channel: 12
[DBG ][m6Thread]: channel page: 0
[DBG ][m6Thread]: channel mask: 134215680
[DBG ][m6Thread]: Mesh prefix: fd:00:0d:b8:00:00:00:00
[DBG ][m6Thread]: Master key: 00:11:22:33:44:55:66:77:88:99:aa:bb:cc:dd:ee:ff
[DBG ][m6Thread]: PSKc: 00:11:22:33:44:55:66:77:88:99:aa:bb:cc:dd:ee:ff
[DBG ][m6Thread]: PSKd: Secret password
[DBG ][ThSA]: service init interface 1, port 61631, options 0
[DBG ][ThSA]: service tasklet init
[DBG ][ThSA]: service init interface 1, port 49191, options 130
[DBG ][ThSA]: Service 2, Uri registration uri: c/tx
[DBG ][ThSA]: Service 2, Uri registration uri: c/mg
[DBG ][ThSA]: Service 2, Uri registration uri: c/ag
[DBG ][ThSA]: Service 2, Uri registration uri: c/pg
[DBG ][ThSA]: Service 2, Uri registration uri: c/cg
[DBG ][ThSA]: Service 2, Uri registration uri: c/as
[DBG ][ThSA]: Service 2, Uri registration uri: c/ps
[DBG ][ThSA]: Service 2, Uri registration uri: c/cs
[DBG ][ThSA]: Service 2, Uri registration uri: c/cp
[DBG ][ThSA]: Service 2, Uri registration uri: c/ca
[DBG ][ThSA]: service init interface 1, port 61631, options 0
[DBG ][ThSA]: Service 3, Uri registration uri: c/mg
[DBG ][ThSA]: Service 3, Uri registration uri: c/ag
[DBG ][ThSA]: Service 3, Uri registration uri: c/pg
[DBG ][ThSA]: Service 3, Uri registration uri: c/cg
[DBG ][ThSA]: Service 3, Uri registration uri: c/pq
[DBG ][ThSA]: Service 3, Uri registration uri: c/es
[DBG ][ThSA]: Service 3, Uri registration uri: c/ab
[DBG ][ThSA]: service init interface 1, port 61631, options 0
[DBG ][ThSA]: Service 4, Uri registration uri: d/dg
[DBG ][ThSA]: Service 4, Uri registration uri: d/dr
[DBG ][ThSA]: Service 4, Uri registration uri: d/dq
[DBG ][ThSA]: service init interface 1, port 61631, options 0
[INFO][m6Thread]: Start Thread bootstrap (Router mode)
[DBG ][ThSA]: service tasklet initialised
[DBG ][nnvm]: ns_nvm_data_read() key=com.arm.nanostack.thread.static_link_cfg, len=120, ctx=0x200057e4
[DBG ][rnvm]: platform_nvm_read() com.arm.nanostack.thread.static_link_cfg len=120
Connection failed!
```

@SeppoTakalo

SeppoTakalo, 2016-12-12:
@mikter @artokin Any ideas?

SeppoTakalo, 2016-12-12:
I can confirm this. Tested on GCC_ARM.
```
$$$07960221002E69383F6AFF93
Connecting...
[DBG ][nsth]: connect()
[DBG ][evlp]: event_loop_thread
[DBG ][m6Thread]: PANID defa
[DBG ][m6Thread]: channel: 18
[DBG ][m6Thread]: channel page: 0
[DBG ][m6Thread]: channel mask: 134215680
[DBG ][m6Thread]: Mesh prefix: fd:00:0d:b8:00:00:00:00
[DBG ][m6Thread]: Master key: 00:11:22:33:44:55:66:77:88:99:aa:bb:cc:dd:ee:ff
[DBG ][m6Thread]: PSKc: 00:11:22:33:44:55:66:77:88:99:aa:bb:cc:dd:ee:ff
[DBG ][m6Thread]: PSKd: Secret password
[DBG ][ThSA]: service init interface 1, port 61631, options 0
[DBG ][ThSA]: service tasklet init
[DBG ][ThSA]: service init interface 1, port 49191, options 130
[DBG ][ThSA]: Service 2, Uri registration uri: c/tx
[DBG ][ThSA]: Service 2, Uri registration uri: c/mg
[DBG ][ThSA]: Service 2, Uri registration uri: c/ag
[DBG ][ThSA]: Service 2, Uri registration uri: c/pg
[DBG ][ThSA]: Service 2, Uri registration uri: c/cg
[DBG ][ThSA]: Service 2, Uri registration uri: c/as
[DBG ][ThSA]: Service 2, Uri registration uri: c/ps
[DBG ][ThSA]: Service 2, Uri registration uri: c/cs
[DBG ][ThSA]: Service 2, Uri registration uri: c/cp
[DBG ][ThSA]: Service 2, Uri registration uri: c/ca
[DBG ][ThSA]: service init interface 1, port 61631, options 0
[DBG ][ThSA]: Service 3, Uri registration uri: c/mg
[DBG ][ThSA]: Service 3, Uri registration uri: c/ag
[DBG ][ThSA]: Service 3, Uri registration uri: c/pg
[DBG ][ThSA]: Service 3, Uri registration uri: c/cg
[DBG ][ThSA]: Service 3, Uri registration uri: c/pq
[DBG ][ThSA]: Service 3, Uri registration uri: c/es
[DBG ][ThSA]: Service 3, Uri registration uri: c/ab
[DBG ][ThSA]: service init interface 1, port 61631, options 0
[DBG ][ThSA]: Service 4, Uri registration uri: d/dg
[DBG ][ThSA]: Service 4, Uri registration uri: d/dr
[DBG ][ThSA]: Service 4, Uri registration uri: d/dq
[DBG ][ThSA]: service init interface 1, port 61631, options 0
[INFO][m6Thread]: Start Thread bootstrap (Router mode)
[DBG ][ThSA]: service tasklet initialised
[DBG ][nnvm]: ns_nvm_data_read() key=com.arm.nanostack.thread.static_link_cfg, len=120, ctx=0x200057e4
[DBG ][rnvm]: platform_nvm_read() com.arm.nanostack.thread.static_link_cfg len=120
Connection failed!
```

SeppoTakalo, 2016-12-12:
When memory changed from 10kB to 16kB it connects:
```
[INFO][m6Thread]: Thread bootstrap ready
[DBG ][m6Thread]: GP IPv6: fd00:db8::ff:fe00:0
[DBG ][m6Thread]: MAC 16-bit: 00:00
[DBG ][m6Thread]: PAN ID: de:fa
[DBG ][m6Thread]: MAC 64-bit: fe:c2:3d:00:00:04:d1:cf
[DBG ][m6Thread]: IID (Based on MAC 64-bit address): fc:c2:3d:00:00:04:d1:cf
[DBG ][nsth]: getOwnIpAddress()
[DBG ][nsth]: getOwnIpAddress()
connected. IP = fd00:db8::ff:fe00:0
````

SeppoTakalo, 2016-12-12:
Fixed in https://github.com/ARMmbed/mbed-os-example-mesh-minimal/commit/b7d25443a4fb454656685d762c4322882ad09c9c

Changed default memory size to 14kB so works in both 6LoWPAN and Thread.

commit b7d25443 (2016-12-12): Change default memory size to 14kB.

Fixes #37
closed 2016-12-12 by SeppoTakalo