ARMmbed/mbed-os-example-mesh-minimal #52: [OoB_testing_5.4] : Thread connection fails on NUCLEO_F429ZI + AT233 shield.

ashok-rao, 2017-02-28:
Steps to reproduce:
1. Clone the example, switch to OoB_5.4 branch, mbed update.
2. Change settings in mbed_app.json. (enabled trace, thread config as below)
3. Update firmware of gateway with https://github.com/ARMmbed/mbed-os-example-client/tree/master/GW_Binary . Tried with both ThreadDynamic.bin and ThreadStatic.bin
4. Compile application: mbed compile -t ARM -m NUCLEO_F429ZI -c
5. Flash binary to target. Output as follows on serial terminal:
```
Connecting...
[DBG ][evlp]: event_loop_thread
Connection failed!
```
Here is mbed_app.json
```
{
    "config": {
        "trace": {
        	"help": "options are true, false",
        	"value": true
        },
        "radio-type":{
            "help": "options are ATMEL, MCR20, NCS36510",
            "value": "ATMEL"
        },
        "mesh-type":{
            "help": "options are MESH_LOWPAN, MESH_THREAD",
            "value": "MESH_THREAD"
        }
    },
    "target_overrides": {
        "*": {
            "target.features_add": ["NANOSTACK", "THREAD_ROUTER", "COMMON_PAL"],
            "mbed-mesh-api.thread-config-panid": "0x0700",
            "mbed-mesh-api.thread-master-key": "{0x10, 0x11, 0x22, 0x33, 0x44, 0x55, 0x66, 0x77, 0x88, 0x99, 0xaa, 0xbb, 0xcc, 0xdd, 0xee, 0xff}",
            "mbed-mesh-api.thread-config-channel": 22,
			"mbed-mesh-api.thread-device-type": "MESH_DEVICE_TYPE_THREAD_ROUTER",
			"mbed-mesh-api.heap-size": 14000,
            "mbed-trace.enable": true,
            "platform.stdio-convert-newlines": true,
            "platform.stdio-baud-rate": 115200
        }
    }
}
```
And here is mbed ls:
```
mbed-os-example-mesh-minimal (e017eb6fe4e3)
|- atmel-rf-driver (57f22763f4d3)
|- mbed-os (3a27568a505b)
`- mcr20a-rf-driver (d8810e105d7d)
```
@MarceloSalazar @SeppoTakalo

ashok-rao, 2017-02-28:
Issue also seen with GCC_ARM and IAR compilers too.

SeppoTakalo, 2017-02-28:
@mikter @artokin Please check.

SeppoTakalo, 2017-02-28:
@ashok-rao Ah.. Actually, you just told what the problem is.
I don't think the bootstrap continues without IPv6.

Router waits to receive ICMPv6 Router-Adverticement message from Ethernet segment. Then that prefix is cloned into mesh network side, and RF side is started up.

If your test environment does not have IPv6 router, then the router newer starts up.

ashok-rao, 2017-02-28:
@SeppoTakalo @mikter @artokin .. I have a working IPv6 tunnel now. Results are still the same unfortunately. Can you please have a look? Screenshot below. mbed_app.json is the same as shown above.

[image]

SeppoTakalo, 2017-02-28:
I cannot replicate this.

I'm using your `mbed_app.json` and it still works

```
Link: mbed-os-example-mesh-minimal
Elf2Bin: mbed-os-example-mesh-minimal
+-----------------------------+--------+-------+-------+
| Module                      |  .text | .data |  .bss |
+-----------------------------+--------+-------+-------+
| Fill                        |    321 |    47 |    48 |
| Misc                        | 233561 |  2634 |  1691 |
| drivers                     |   2249 |     4 |   164 |
| features/FEATURE_COMMON_PAL |  14317 |    94 |  8431 |
| features/FEATURE_LWIP       |     44 |     0 | 12536 |
| features/mbedtls            | 119141 |    51 |  8867 |
| features/nanostack          |  16618 |    37 | 14134 |
| features/netsocket          |   2819 |    85 |     0 |
| hal                         |    622 |     0 |     8 |
| platform                    |   1742 |    20 |   299 |
| rtos                        |    820 |     4 |     4 |
| rtos/rtx                    |   7347 |    20 |  6870 |
| targets/TARGET_STM          |  17840 |     4 |  1292 |
| Subtotals                   | 417441 |  3000 | 54344 |
+-----------------------------+--------+-------+-------+
Allocated Heap: 24576 bytes
Allocated Stack: unknown
Total Static RAM memory (data + bss): 57344 bytes
Total RAM memory (data + bss + heap + stack): 81920 bytes
Total Flash memory (text + data + misc): 420441 bytes

Image: ./BUILD/NUCLEO_F429ZI/GCC_ARM/mbed-os-example-mesh-minimal.bin
~/src/tmp/mbed-os-example-mesh-minimal$ cat mbed_app.json 
{
    "config": {
        "trace": {
            "help": "options are true, false",
            "value": true
        },
        "radio-type":{
            "help": "options are ATMEL, MCR20, NCS36510",
            "value": "ATMEL"
        },
        "mesh-type":{
            "help": "options are MESH_LOWPAN, MESH_THREAD",
            "value": "MESH_THREAD"
        }
    },
    "target_overrides": {
        "*": {
            "target.features_add": ["NANOSTACK", "THREAD_ROUTER", "COMMON_PAL"],
            "mbed-mesh-api.thread-config-panid": "0x0700",
            "mbed-mesh-api.thread-master-key": "{0x10, 0x11, 0x22, 0x33, 0x44, 0x55, 0x66, 0x77, 0x88, 0x99, 0xaa, 0xbb, 0xcc, 0xdd, 0xee, 0xff}",
            "mbed-mesh-api.thread-config-channel": 22,
            "mbed-mesh-api.thread-device-type": "MESH_DEVICE_TYPE_THREAD_ROUTER",
            "mbed-mesh-api.heap-size": 14000,
            "mbed-trace.enable": true,
            "platform.stdio-convert-newlines": true,
            "platform.stdio-baud-rate": 115200
        }
    }
}
~/src/tmp/mbed-os-example-mesh-minimal$ mbed ls
mbed-os-example-mesh-minimal (e017eb6fe4e3)
|- atmel-rf-driver (57f22763f4d3)
|- mbed-os (3a27568a505b)
`- mcr20a-rf-driver (d8810e105d7d)
```

After flashing
```
[DBG ][ThSA]: Service 3, Uri registration uri: a/as
[DBG ][ThSA]: Service 3, Uri registration uri: a/ar
[DBG ][ThSA]: Service 3, Uri registration uri: c/lp
[DBG ][ThSA]: Service 3, Uri registration uri: c/la
[DBG ][ThSA]: Service 3, Uri registration uri: a/sd
[DBG ][ThSA]: Service 3, Uri registration uri: c/as
[DBG ][ThSA]: Service 3, Uri registration uri: c/ps
[DBG ][ThSA]: Service 3, Uri registration uri: c/cs
[DBG ][ThSA]: service init interface 1, port 61631, options 0
[DBG ][ThSA]: Service 6, Uri registration uri: a/aq
[DBG ][ThSA]: service init interface 1, port 61631, options 0
[DBG ][ThSA]: Service 7, Uri registration uri: a/an
[DBG ][ThSA]: Service 7, Uri registration uri: a/ae
[DBG ][m6Thread]: app_parse_network_event() 0
[INFO][m6Thread]: Thread bootstrap ready
[DBG ][m6Thread]: GP IPv6: fd00:db8::ff:fe00:0
[DBG ][m6Thread]: MAC 16-bit: 00:00
[DBG ][m6Thread]: PAN ID: 07:00
[DBG ][m6Thread]: MAC 64-bit: fe:c2:3d:00:00:04:d1:cf
[DBG ][m6Thread]: IID (Based on MAC 64-bit address): fc:c2:3d:00:00:04:d1:cf
[DBG ][nsth]: getOwnIpAddress()
[DBG ][nsth]: getOwnIpAddress()
connected. IP = fd00:db8::ff:fe00:0
```

ashok-rao, 2017-02-28:
Issue resolved. On my target board (NUCLEO_F429ZI), I had the zero resistor manually shorted to SB122 (SPI pin clash with ethernet controller's CRS_DV pin for a different application) instead of the factory config at SB121 .. which may have interfered with the SPI config for the 6LoWPAN shield.
Closing issue now.

SeppoTakalo, 2017-02-28:
Hi @ashok-rao
Can you also try using binaries from 5.4 RC/OOB branch: https://github.com/ARMmbed/mbed-os-example-client/tree/oob_test_mbed-os-5.4/GW_Binary

NOTE: I believe actually the `-static` one should be able to boot, even when you don't have proper IPv6 network. Just connect the ethernet cable to somewhere.

SeppoTakalo, 2017-02-28:
Ah, ok.

Thanks for reporting.

ashok-rao, 2017-02-28:
@SeppoTakalo , I tried the 5.4 GW_Binary you mentioned above, I'm getting these warnings after that on the gateway and unable to connect the target.
```
IP Addr: 2a01:348:6:8810::
LL Addr: (STALE 0) b8:27:eb:e6:9f:d4
Router IP Addr: fe80::3f91:4b36:f231:95bb
LL Addr: (STALE 0) b8:27:eb:e6:9f:d4
Neighbour Cache 0
Reachable Time: 36672   Retrans Timer: 1000   MTU: 1280
[DBG ][icmp]: Possible Default Router
[DBG ][ip6s]: Interface ID: 1, ipv6: 2a01:348:6:8810:fec2:3dff:fe04:8a99
[DBG ][6Bo ]: Send MLE Advertisement
[WARN][mDa ]: Unknown neighbour (fe:c2:3d:00:00:04:e4:a3)
[WARN][mDa ]: Duplicate Frame counter: -1
[DBG ][icmp]: Possible Default Router
[DBG ][ip6s]: Interface ID: 1, ipv6: 2a01:348:6:8810:fec2:3dff:fe04:8a99
[WARN][mDa ]: Unknown neighbour (fe:c2:3d:00:00:04:e4:a3)
[WARN][mDa ]: Duplicate Frame counter: -1
[WARN][mDa ]: Unknown neighbour (fe:c2:3d:00:00:04:e4:a3)
[WARN][mDa ]: Duplicate Frame counter: -1
[DBG ][icmp]: Possible Default Router
[DBG ][ip6s]: Interface ID: 1, ipv6: 2a01:348:6:8810:fec2:3dff:fe04:8a99

```

SeppoTakalo, 2017-03-01:
At least you were able to get IPv6 address from the router `2a01:348:6:8810:fec2:3dff:fe04:8a99`

What exactly did not work?

How far away the devices are? Duplicate frame counter might mean transmission errors.

Also, Thread GW images were updated just a minutes ago. They noticed a bug that if Border Router is not the first device to start up, it might take 10 minutes to change mode to router.

Try again by shutting down all devices, then start up the BR, then other devices.

ashok-rao, 2017-03-01:
Yes, got the IPv6 address from the GW. The connection from end node failed. The GW and end node are sitting on the same table..maybe few inches away :-). I will try with the new GW binaries & report back.

SeppoTakalo, 2017-06-14:
Closing this as there has not been enough proof that any issues still remaining.

F429CI is part of our CI testing and this problems is not visible there: https://jenkins-internal.mbed.com/job/mbed-os-example-mesh-minimal-6lowpan-hw-test/

closed 2017-06-14 by SeppoTakalo