ARMmbed/mbed-os-example-mesh-minimal #53: Compile fail with make_iar for NUCLEO_F42ZI.

ashok-rao, 2017-02-28:
Steps to reproduce:
1. mbed export -i make_iar -m NUCLEO_F429ZI
2. make clean
3. make

Results in compilation error as below:
```
C:\Ashok\mbed_CLI\v5_4_OoB\mbed-os-example-mesh-minimal>mbed export -i make_iar -m NUCLEO_F429ZI
c:\python27\lib\site-packages\fuzzywuzzy\fuzz.py:35: UserWarning: Using slow pure-python SequenceMatcher. Install python-Levenshtein to remove this warning
  warnings.warn('Using slow pure-python SequenceMatcher. Install python-Levenshtein to remove this warning')
Scan: .
Scan: FEATURE_BLE
Scan: FEATURE_COMMON_PAL
Scan: FEATURE_LWIP
Scan: FEATURE_UVISOR
Scan: FEATURE_ETHERNET_HOST
Scan: FEATURE_LOWPAN_BORDER_ROUTER
Scan: FEATURE_LOWPAN_HOST
Scan: FEATURE_LOWPAN_ROUTER
Scan: FEATURE_NANOSTACK
Scan: FEATURE_NANOSTACK_FULL
Scan: FEATURE_THREAD_BORDER_ROUTER
Scan: FEATURE_THREAD_END_DEVICE
Scan: FEATURE_THREAD_ROUTER
Scan: FEATURE_STORAGE

C:\Ashok\mbed_CLI\v5_4_OoB\mbed-os-example-mesh-minimal>make clean
rmdir /S /Q "BUILD"

C:\Ashok\mbed_CLI\v5_4_OoB\mbed-os-example-mesh-minimal>make
C:/Ashok/mbed_CLI/v5_4_OoB/mbed-os-example-mesh-minimal/makefile:726: warning: overriding commands for target `.s.o'
C:/Ashok/mbed_CLI/v5_4_OoB/mbed-os-example-mesh-minimal/makefile:721: warning: ignoring old commands for target `.s.o'
"Compile: NanostackRfPhyAtmel.cpp"
process_begin: CreateProcess(NULL, iccarm --no_wrap_diagnostics -e --diag_suppress=Pa050,Pa084,Pa093,Pa082 -On -r --cpu Cortex-M4F --thumb --dlib_config DLib_Config_Full.h --c++ --no_rtti --no_exceptions --guard_calls --no_static_destruction --guard_calls --no_static_destruction -DFEATURE_NANOSTACK=
1 -DDEVICE_SPISLAVE=1 -DFEATURE_LWIP=1 -D__MBED__=1 -DDEVICE_I2CSLAVE=1 -D__FPU_PRESENT=1 -DTRANSACTION_QUEUE_SIZE_SPI=2 -DUSBHOST_OTHER -DDEVICE_PORTINOUT=1 -DTARGET_RTOS_M4_M7 -DTARGET_NUCLEO_F429ZI -DMBED_BUILD_TIMESTAMP=1488309066.45 -DDEVICE_RTC=1 -DTOOLCHAIN_object -DTARGET_STM32F4 -D__CMSIS_R
TOS -DTARGET_FF_ARDUINO -DTARGET_FLASH_CMSIS_ALGO -DDEVICE_CAN=1 -DTARGET_CORTEX_M -DTARGET_DEBUG -DDEVICE_I2C_ASYNCH=1 -DTARGET_LIKE_CORTEX_M4 -DDEVICE_ANALOGOUT=1 -DTARGET_M4 -DTARGET_UVISOR_UNSUPPORTED -DDEVICE_PORTOUT=1 -DDEVICE_SPI_ASYNCH=1 -DTOOLCHAIN_IAR -DDEVICE_INTERRUPTIN=1 -DTARGET_STM32F
429xx -DDEVICE_I2C=1 -DTARGET_STM32F429 -D__CORTEX_M4 -DDEVICE_STDIO_MESSAGES=1 -DDEVICE_PWMOUT=1 -DTARGET_LIKE_MBED -DDEVICE_LOWPOWERTIMER=1 -DFEATURE_COMMON_PAL=1 -DDEVICE_PORTIN=1 -DTARGET_STM -DDEVICE_SERIAL_FC=1 -DFEATURE_THREAD_END_DEVICE=1 -DDEVICE_TRNG=1 -DTARGET_STM32F429ZI -D__MBED_CMSIS_R
TOS_CM -DDEVICE_SLEEP=1 -DDEVICE_SPI=1 -DUSB_STM_HAL -DDEVICE_ERROR_RED=1 -DTARGET_STM32F429xI -DDEVICE_ANALOGIN=1 -DDEVICE_SERIAL=1 -DDEVICE_FLASH=1 -DARM_MATH_CM4 --preinclude=mbed_config.h -I../ -I../. -I.././DebugConfig -I.././atmel-rf-driver -I.././atmel-rf-driver/atmel-rf-driver -I.././atmel-r
f-driver/source -I.././configs -I.././mbed-os -I.././mbed-os/cmsis -I.././mbed-os/cmsis/TOOLCHAIN_IAR -I.././mbed-os/drivers -I.././mbed-os/events -I.././mbed-os/events/equeue -I.././mbed-os/features -I.././mbed-os/features/FEATURE_COMMON_PAL -I.././mbed-os/features/FEATURE_COMMON_PAL/mbed-client-ra
ndlib -I.././mbed-os/features/FEATURE_COMMON_PAL/mbed-client-randlib/mbed-client-randlib -I.././mbed-os/features/FEATURE_COMMON_PAL/mbed-client-randlib/mbed-client-randlib/platform -I.././mbed-os/features/FEATURE_COMMON_PAL/mbed-client-randlib/source -I.././mbed-os/features/FEATURE_COMMON_PAL/mbed-c
oap -I.././mbed-os/features/FEATURE_COMMON_PAL/mbed-coap/doxygen -I.././mbed-os/features/FEATURE_COMMON_PAL/mbed-coap/mbed-coap -I.././mbed-os/features/FEATURE_COMMON_PAL/mbed-coap/source -I.././mbed-os/features/FEATURE_COMMON_PAL/mbed-coap/source/include -I.././mbed-os/features/FEATURE_COMMON_PAL/m
bed-trace -I.././mbed-os/features/FEATURE_COMMON_PAL/mbed-trace/mbed-trace -I.././mbed-os/features/FEATURE_COMMON_PAL/mbed-trace/source -I.././mbed-os/features/FEATURE_COMMON_PAL/nanostack-hal-mbed-cmsis-rtos -I.././mbed-os/features/FEATURE_COMMON_PAL/nanostack-hal-mbed-cmsis-rtos/cs_nvm -I.././mbed
-os/features/FEATURE_COMMON_PAL/nanostack-hal-mbed-cmsis-rtos/nvm -I.././mbed-os/features/FEATURE_COMMON_PAL/nanostack-libservice -I.././mbed-os/features/FEATURE_COMMON_PAL/nanostack-libservice/mbed-client-libservice -I.././mbed-os/features/FEATURE_COMMON_PAL/nanostack-libservice/mbed-client-libserv
ice/platform -I.././mbed-os/features/FEATURE_COMMON_PAL/nanostack-libservice/source -I.././mbed-os/features/FEATURE_COMMON_PAL/nanostack-libservice/source/IPv6_fcf_lib -I.././mbed-os/features/FEATURE_COMMON_PAL/nanostack-libservice/source/libBits -I.././mbed-os/features/FEATURE_COMMON_PAL/nanostack-
libservice/source/libList -I.././mbed-os/features/FEATURE_COMMON_PAL/nanostack-libservice/source/libTrace -I.././mbed-os/features/FEATURE_COMMON_PAL/nanostack-libservice/source/libTrace/scripts -I.././mbed-os/features/FEATURE_COMMON_PAL/nanostack-libservice/source/libip6string -I.././mbed-os/feature
s/FEATURE_COMMON_PAL/nanostack-libservice/source/nsdynmemLIB -I.././mbed-os/features/FEATURE_COMMON_PAL/nanostack-libservice/source/nvmHelper -I.././mbed-os/features/FEATURE_COMMON_PAL/sal-stack-nanostack-eventloop -I.././mbed-os/features/FEATURE_COMMON_PAL/sal-stack-nanostack-eventloop/nanostack-ev
ent-loop -I.././mbed-os/features/FEATURE_COMMON_PAL/sal-stack-nanostack-eventloop/nanostack-event-loop/platform -I.././mbed-os/features/FEATURE_COMMON_PAL/sal-stack-nanostack-eventloop/source -I.././mbed-os/features/FEATURE_LWIP -I.././mbed-os/features/FEATURE_LWIP/lwip-interface -I.././mbed-os/feat
ures/FEATURE_LWIP/lwip-interface/lwip -I.././mbed-os/features/FEATURE_LWIP/lwip-interface/lwip-eth -I.././mbed-os/features/FEATURE_LWIP/lwip-interface/lwip-eth/arch -I.././mbed-os/features/FEATURE_LWIP/lwip-interface/lwip-eth/arch/TARGET_STM -I.././mbed-os/features/FEATURE_LWIP/lwip-interface/lwip-e
th/arch/TARGET_STM/TARGET_STM32F4 -I.././mbed-os/features/FEATURE_LWIP/lwip-interface/lwip-eth/arch/TARGET_STM/TARGET_STM32F4/TARGET_NUCLEO_F429ZI -I.././mbed-os/features/FEATURE_LWIP/lwip-interface/lwip-sys -I.././mbed-os/features/FEATURE_LWIP/lwip-interface/lwip-sys/arch -I.././mbed-os/features/FE
ATURE_LWIP/lwip-interface/lwip/src -I.././mbed-os/features/FEATURE_LWIP/lwip-interface/lwip/src/api -I.././mbed-os/features/FEATURE_LWIP/lwip-interface/lwip/src/core -I.././mbed-os/features/FEATURE_LWIP/lwip-interface/lwip/src/core/ipv4 -I.././mbed-os/features/FEATURE_LWIP/lwip-interface/lwip/src/co
re/ipv6 -I.././mbed-os/features/FEATURE_LWIP/lwip-interface/lwip/src/include -I.././mbed-os/features/FEATURE_LWIP/lwip-interface/lwip/src/include/lwip -I.././mbed-os/features/FEATURE_LWIP/lwip-interface/lwip/src/include/lwip/priv -I.././mbed-os/features/FEATURE_LWIP/lwip-interface/lwip/src/include/l
wip/prot -I.././mbed-os/features/FEATURE_LWIP/lwip-interface/lwip/src/include/netif -I.././mbed-os/features/FEATURE_LWIP/lwip-interface/lwip/src/include/netif/ppp -I.././mbed-os/features/FEATURE_LWIP/lwip-interface/lwip/src/include/netif/ppp/polarssl -I.././mbed-os/features/FEATURE_LWIP/lwip-interfa
ce/lwip/src/netif -I.././mbed-os/features/filesystem -I.././mbed-os/features/filesystem/bd -I.././mbed-os/features/filesystem/fat -I.././mbed-os/features/filesystem/fat/ChaN -I.././mbed-os/features/frameworks -I.././mbed-os/features/frameworks/greentea-client -I.././mbed-os/features/frameworks/green
tea-client/greentea-client -I.././mbed-os/features/frameworks/greentea-client/source -I.././mbed-os/features/frameworks/unity -I.././mbed-os/features/frameworks/unity/source -I.././mbed-os/features/frameworks/unity/unity -I.././mbed-os/features/frameworks/utest -I.././mbed-os/features/frameworks/ute
st/source -I.././mbed-os/features/frameworks/utest/utest -I.././mbed-os/features/mbedtls -I.././mbed-os/features/mbedtls/importer -I.././mbed-os/features/mbedtls/inc -I.././mbed-os/features/mbedtls/inc/mbedtls -I.././mbed-os/features/mbedtls/platform -I.././mbed-os/features/mbedtls/platform/inc -I..
/./mbed-os/features/mbedtls/platform/src -I.././mbed-os/features/mbedtls/src -I.././mbed-os/features/nanostack -I.././mbed-os/features/nanostack/FEATURE_NANOSTACK -I.././mbed-os/features/nanostack/FEATURE_NANOSTACK/coap-service -I.././mbed-os/features/nanostack/FEATURE_NANOSTACK/coap-service/coap-se
rvice -I.././mbed-os/features/nanostack/FEATURE_NANOSTACK/coap-service/source -I.././mbed-os/features/nanostack/FEATURE_NANOSTACK/coap-service/source/include -I.././mbed-os/features/nanostack/FEATURE_NANOSTACK/mbed-mesh-api -I.././mbed-os/features/nanostack/FEATURE_NANOSTACK/mbed-mesh- [... cut ...] /./mbed-os/targets/TARGET_STM/TARGET_STM32F4/TARGET_ST
M32F429xI/device/TOOLCHAIN_IAR -I.././mbed-os/targets/TARGET_STM/TARGET_STM32F4/device -I.././mcr20a-rf-driver -I.././mcr20a-rf-driver/mcr20a-rf-driver -I.././mcr20a-rf-driver/source -o atmel-rf-driver/source/NanostackRfPhyAtmel.o ../atmel-rf-driver/source/NanostackRfPhyAtmel.cpp, ...) failed.
make (e=2): The system cannot find the file specified.
make[1]: *** [atmel-rf-driver/source/NanostackRfPhyAtmel.o] Error 2
make: *** [all] Error 2
```

SeppoTakalo, 2017-03-01:
I cannot replicate this.

*Tested with K64F:*

|           Tool |Result |
|----------------|-------|
|     make_armc5 |  PASS |
|   make_gcc_arm |  PASS |
|       make_iar |  PASS |
|       uvision5 |  PASS |
|            iar |  PASS |

*Tested with Nucleo F429ZI:*

```

$ mbed export -i make_iar -m NUCLEO_F429ZI
$ make clean
$ make

...

'ielftool' --ihex mbed-os-example-mesh-minimal.elf mbed-os-example-mesh-minimal.hex

   IAR ELF Tool V9.20.15.161 [BUILT at IAR]
   Copyright 2007-2016 IAR Systems AB.

Loading mbed-os-example-mesh-minimal.elf
Saving ihex file to mbed-os-example-mesh-minimal.hex
```

Does any of the Make commands work for you?
The error code `make (e=2): The system cannot find the file specified.` looks like its coming from `process_begin: CreateProcess(NULL, iccarm ...`

Can you try `iccarm --version`
```
$ iccarm --version
IAR ANSI C/C++ Compiler V7.70.2.11706/W32 for ARM
```

If its not in the path, it cannot work.

ashok-rao, 2017-03-01:
Yes, my PATH variables were messed up for some reason. Issue resolved. Thanks @SeppoTakalo.

closed 2017-03-01 by SeppoTakalo