ARMmbed/mbed-os-example-mesh-minimal #55: [OoB_testing_5.4]:ublox ODIN W2, 6LoWPAN connection failure with MCR20A.

ashok-rao, 2017-03-01:
Steps to reproduce:
1. mbed compile -t ARM -m ublox_evk_odin_w2 -c
2. Flash binary to target.
3. Update firmware of GW with gateway6LoWPANStatic.bin / gateway6LoWPANDynamic.bin from here:https://github.com/ARMmbed/mbed-os-example-client/tree/master/GW_Binary
4. Make sure GW is up & running before powering on the end node.
5. Once IPv6 address is obtained on the GW, power on the target.

Nothing happens at the end node. Waits for a long time as below:
```
Connecting...
[DBG ][core]: Allocate Root Tasklet
[DBG ][6lo ]: P.Init

[DBG ][evlp]: event_loop_thread
[DBG ][core]: NS Root task Init
[DBG ][sck ]: Socket Tasklet Generated
```
Below is serial terminal output from GW:
```
<snip>
Router IP Addr: fe80::6320:f661:71fd:98d3
LL Addr: (STALE 0) b8:27:eb:e6:9f:d4
Neighbour Cache 0
Reachable Time: 33373   Retrans Timer: 1000   MTU: 1280
[DBG ][icmp]: Possible Default Router
[DBG ][ip6s]: Interface ID: 1, ipv6: 2a01:348:6:8810:fec2:3dff:fe04:8a99
[DBG ][rplc]: transmit dio, rank: 80
[DBG ][icmp]: Possible Default Router
[DBG ][ip6s]: Interface ID: 1, ipv6: 2a01:348:6:8810:fec2:3dff:fe04:8a99
</snip>
```
mbed_app.json is below:
```
{
    "config": {
        "trace": {
        	"help": "options are true, false",
        	"value": true
        },
        "radio-type":{
            "help": "options are ATMEL, MCR20, NCS36510",
            "value": "MCR20"
        },
        "mesh-type":{
            "help": "options are MESH_LOWPAN, MESH_THREAD",
            "value": "MESH_LOWPAN"
        }
    },
    "target_overrides": {
        "*": {
            "target.features_add": ["NANOSTACK", "LOWPAN_ROUTER", "COMMON_PAL"],
			"mbed-mesh-api.6lowpan-nd-panid-filter": "0xffff",
			"mbed-mesh-api.6lowpan-nd-channel-page": 0,
			"mbed-mesh-api.6lowpan-nd-channel": 12,
			"mbed-mesh-api.6lowpan-nd-channel-mask": "(1<<12)",
			"mbed-mesh-api.6lowpan-nd-device-type": "NET_6LOWPAN_ROUTER",
			"mbed-mesh-api.heap-size": 14000,
            "mbed-trace.enable": true,
            "platform.stdio-convert-newlines": true,
            "platform.stdio-baud-rate": 115200
        }
    }
}
```

SeppoTakalo, 2017-03-01:
FRDM-CR20A does not work with Ublox ODIN W2.
Atmel shield works with this board.

I believe it is because it uses `UART1 RX` pin in same pin as `MCR20A_SPI_IRQ`which is `D2`
[image]

See https://github.com/ARMmbed/mcr20a-rf-driver/blob/master/mcr20a-rf-driver/NanostackRfPhyMcr20a.h#L40

SeppoTakalo, 2017-03-08:
Documented in https://github.com/ARMmbed/mbed-os-example-mesh-minimal/blob/oob_test_mbed-os-5.4/Hardware.md

SeppoTakalo, 2017-03-09:
Closed the issue as it is now documented and documentation points also here.

closed 2017-03-09 by SeppoTakalo