ARMmbed/mbed-os-example-mesh-minimal #89: Cannot compile on Online Compiler

brandonboesch, 2017-06-09:
Using the online compiler and selecting the FRDM-KW24D development board, I could not compile the mesh-minimal example on branch mbed-os-5.5.0-rc1-oob. For this example, I changed the mbed_app.json file so that it is targeting the KW24D, and setting everything in Thread mode.  When compiling, the following errors occur:

[image]

maclobdell, 2017-06-14:
This platform is not enabled in the online compiler yet.

maclobdell, 2017-06-14:
Actually, I think this should still have worked.  We should investigate it.  

By the way, since there is so much dependency on the board you are using, I highly recommend that when you create an issue, put the name of the board in the description.  for example "Cannot compile on Online Compiler for FRDM-KW24D" is much clearer and less alarming.  Also, in this case, if it didn't work in the online compiler, it would be helpful to know if you tried another program, such as mbed-os-example-blinky.  Also it would be helpful to know if this example worked in the online compiler with another platform such as FRDM-K64F.

SeppoTakalo, 2017-06-14:
@ashok-rao @MarceloSalazar @maclobdell Are they any instructions somewhere that how can I specify which branch to pull into the online compiler and which mbed OS version to check out?

ashok-rao, 2017-06-14:
@SeppoTakalo @brandonboesch .. from the OoB spreadsheet:

[image]

SeppoTakalo, 2017-06-14:
How does it update the `mbed-os` and libaries to point the versions specifed in the `.lib` files?

SeppoTakalo, 2017-06-14:
I cloned the application using Github URL and with Git hash from the RC2 OOB branch and now the build seems to work.

[image]

Closing this.

closed 2017-06-14 by SeppoTakalo

ashok-rao, 2017-06-14:
@SeppoTakalo .. we need to manually switch to the required branch.. you can see from below that I'm currently using rc2 branch..
[image]

SeppoTakalo, 2017-06-14:
This is not the same this as mbed-os revision.

So revision you are pointing is the branch in the application. This is what I already have done.

Then, next step is to update the mbed-os to point to the Git hash that the .lib file specifies.
I don't know how to do it, so only way to achieve it was to clone this application using the import button and URL `https://github.com/ARMmbed/mbed-os-example-mesh-minimal.git#a1213c45b6bde1915fcdbfaf0f0ebb41333e8264`
and check you don't check the tick-box "Update all to latest.." then what you get should be the RC2 branch pointing to mbed OS RC2.

Update latest looks like it can only update to tagged/released versions of libraries, not a branch.

brandonboesch, 2017-06-14:
Ah, I did not realize that I was supposed to use the developer version of the online compiler.  I now have the application running properly when using the correct online compiler.