ARMmbed/mbed-os-example-mesh-minimal #90: [OoB_5_5] Compile warnings

AriParkkila, 2017-06-12:
ARM 5.06u5 log:
[log.txt](https://github.com/ARMmbed/mbed-os-example-mesh-minimal/files/1068213/log.txt)

GCC 6.3.1 log:
[log.txt](https://github.com/ARMmbed/mbed-os-example-mesh-minimal/files/1068222/log.txt)

0xc0170, 2017-06-13:
@AriParkkila Please provide more details. The headline should show what is the issue, build error for this target, or similar. The body of the issue details to reproduce. The logs are nice to have, but would be better if the bug report would contain at least one paragraph with description how to reproduce and what are the symptoms (if runtime error, then how it fails, call stack, code snippet, anything that helps).

AriParkkila, 2017-06-14:
Thanks for response. All compiler warnings should be removed due to each warning is a potential program defect, fails to build with -Wall flags and also excessive (even false positive) warnings may hide any real problems.

The logs contain compiler configuration, as an example NUCLEO_F411RE, ARM:
mbed compile -t GCC_ARM -m NUCLEO_F411RE

I updated the compile commands to issue description.

SeppoTakalo, 2017-06-14:
I agree, there should not be any compiler warnings when building a released code.
We keep this open until warnings are fixed.

SeppoTakalo, 2017-08-01:
Fixed in https://github.com/ARMmbed/mbed-os/pull/4738

closed 2017-08-01 by SeppoTakalo