ARMmbed/mbed-os-example-mesh-minimal #91: [OoB_5_5] Compile errors

AriParkkila, 2017-06-12:
[Log from IAR Systems\Embedded Workbench 8.0](https://github.com/ARMmbed/mbed-os-example-mesh-minimal/files/1068233/iar.txt)

0xc0170, 2017-06-13:
We support 7.8 IAR version at the moment.

AriParkkila, 2017-06-14:
Thanks. Where can I find officially supported mbed toolchains on the mbed website?

SeppoTakalo, 2017-06-14:
Uhm...
https://docs.mbed.com/docs/mbed-os-handbook/en/latest/dev_tools/options/

Specifies IAR 7.7, but for mbed OS 5.5 this is going to change.
I hope that OOB testers are informed.

ashok-rao, 2017-06-14:
The OoB spreadsheet's Introduction tab says to use IAR 7.8 ..

AriParkkila, 2017-06-14:
I probably should have read OoB instructions more carefully :)

SeppoTakalo, 2017-06-14:
Looks like the target is NUCLEO_F411RE and that does not have hardware random number generator.
Looks like it triggers some kind of weird build where part of mbed TLS is enabled and part of it is not.
CoAP service should be able to build without security, and it has been used that way. This needs more investigation why it only appears on this platform.

AriParkkila, 2017-06-14:
I get the same error with IAR 7.8 [log](https://github.com/ARMmbed/mbed-os-example-mesh-minimal/files/1074317/log.txt).

AriParkkila, 2017-06-14:
As expected IAR IDE build gives the same error as mbed compile due to the same compiler. However IDE keeps compiling after the error so it reveals some additional warnings, see 
[log from IAR EW IDE 7.80.1](https://github.com/ARMmbed/mbed-os-example-mesh-minimal/files/1074342/iar.ide.txt).

SeppoTakalo, 2017-06-14:
Can you also provide the `mbed_app.json` used for the build.

AriParkkila, 2017-06-14:
Please see [mbed_app.json](https://github.com/ARMmbed/mbed-os-example-mesh-minimal/files/1074397/mbed_app.json.txt).

SeppoTakalo, 2017-06-14:
I can see now that these are compiler warnings in GCC and ARM, and failures in IAR.

SeppoTakalo, 2017-06-14:
Verified that this example does not compile with NUCLEO_F401RE with IAR.

This needs to be reported in known issues list.

adbridge, 2017-09-18:
@SeppoTakalo Is there any update on this? Has any further investigation been done at all or is the intention just to propagate this to the known issues for 5.6 too ?

SeppoTakalo, 2017-09-18:
No update yet, but I'm verifying this locally now.
I'll update whether the issue still exist.

SeppoTakalo, 2017-09-18:
Problem still exists. Reported to module causing the compilation error: https://github.com/ARMmbed/coap-service/issues/75

Seems to be compilation warning on GCC and ARM, and failure on IAR.

Should be reported to known issue list. Can only be closed once the CoAP service is fixed into Mbed OS master.

SeppoTakalo, 2017-09-28:
Fix merged to master, closing.

closed 2017-09-28 by SeppoTakalo