ARMmbed/mbed-os-example-mesh-minimal #92: Button and LED mapping not included in .json file for KW24D

brandonboesch, 2017-06-12:
Working thought the example with the KW24D, I am able to get the device to connect to an IP address, but I have no way of verifying that communication is made between multiple devices, as mbed_app.json does not have the buttons/LEDs mapped yet.

maclobdell, 2017-06-14:
Thanks for catching this issue.  Try adding this to the target_overrides section of mbed_app.json.  It should work.  Make sure this is set to true also "enable-led-control-example": true,

        "KW24D": {
            "LED": "LED1",
            "BUTTON": "SW1"
        }

If that works, you can make a pull request to add it.  
I just made a similar pull request for another platform.
https://github.com/ARMmbed/mbed-os-example-mesh-minimal/pull/99

brandonboesch, 2017-06-14:
Thanks Mac.  I added that code to mbed_app.json, and successfully communicated between KW24D in thread mode.  I will make a pull request now to add those changes.

SeppoTakalo, 2017-06-15:
Merged in.
Thanks for providing the pin-names and testing this.

closed 2017-06-15 by SeppoTakalo