ARMmbed/mbed-os-example-mesh-minimal #93: Issues with DAPLink

brandonboesch, 2017-06-12:
I am able to get the board to connect to an IP using the DAPLink firmware found here:
\\venus.cambridge.arm.com\Groups\DevSys\mbed\DAPLink\releases\uvision_0243_2f498feb_kw24d\k20dx_frdmkw24d_if_crc_legacy_0x8000.bin

But using the DAPLink firmware referenced here, the program no longer runs:
https://user-images.githubusercontent.com/16062082/27047272-2158ae78-4f6d-11e7-8417-5caa0099159b.png

Have not tested with other DAPLink firmware.

maclobdell, 2017-06-14:
In my opinion, this is not really an issue with this example application, but an issue with the board platform.  Since the newer version of the Daplink firmware works, then it is safe to say it has been resolved.  This platform is not officially mbed Enabled yet and it is not live on developer.mbed.org.  When it goes live we can make sure that it has a link to the latest working firmware.  Thanks!

SeppoTakalo, 2017-06-14:
Closing as not related to this application.

closed 2017-06-14 by SeppoTakalo