ARMmbed/mbed-os-example-mesh-minimal #94: Throwing errors when running uVision's compiled version of program.

brandonboesch, 2017-06-12:
When running after exporting and compiling in uVision, the following error occurs on serial output:

Mutex 20000490 error -6: Not allowed in ISR context

SeppoTakalo, 2017-06-13:
Can you tell me what target board this is, and output of `mbed ls`

SeppoTakalo, 2017-06-14:
Also, can you provide me a call stack from the error condition.
Thanks.

brandonboesch, 2017-06-14:
I am using a KW24D. The output of `mbed ls` is:
[image]

The call stack the moment before the error is thrown is here:
[image]

brandonboesch, 2017-06-14:
I just tried using running the program after exporting and compiling for make_arm5c, and received the same error:
Mutex 1fff8c98 error -6: Not allowed in ISR context

SeppoTakalo, 2017-06-15:
Thanks.
This looks clear now. MCR20A driver is calling Nanostack's API from IRQ context.
We should report this issue there.

ashok-rao, 2017-06-15:
@0xc0170 .. looks like this one is for MCR20 driver authors. Can you please tag relevant people? Thanks.

0xc0170, 2017-06-15:
> @0xc0170 .. looks like this one is for MCR20 driver authors. Can you please tag relevant people? Thanks.

@SeppoTakalo should know better, can you please?

MarceloSalazar, 2017-06-27:
@SeppoTakalo please comment

SeppoTakalo, 2017-09-14:
This was fixed in https://github.com/ARMmbed/mcr20a-rf-driver/commit/72f895683b49bce928fbfaddd7d64ab681122eec

Needs re-testing in Uvision

closed 2017-09-14 by SeppoTakalo

mray19027, 2017-09-27:
Confirmed this works in Uvision5 using exporters