ARMmbed/mbed-os-example-mesh-minimal #95: make_gcc_arm throwing errors when running make

brandonboesch, 2017-06-12:
After exporting with make_gcc_arm, and then running make, the following error occurs:

[image]

SeppoTakalo, 2017-06-13:
Thanks for reporting.
I assume that you have KW24D platform which suffers from small RAM size.
Static allocations cannot be more that 31kB because this device have its RAM split into two equal sized sections. GCC linker can only utilize one block for one section .BSS, .DATA or stack.

Couple of ways to workaround this:
* Use ARM compiler. This will allow .BSS and .DATA to grow from `m_data` to `m_data2` section.
* Wait for RC2. [This](https://github.com/ARMmbed/mbed-os/pull/4493) will drop 14kB of RAM usage
* Follow instructions for [memory optimisations of this application](https://github.com/ARMmbed/mbed-os-example-mesh-minimal/blob/master/README.md#memory-optimizations).

AriParkkila, 2017-06-14:
@brandonboesch The "make error 2" is probably due to missing path to compiler? I had the same error although I have correct paths in mbed config (GCC_ARM_PATH, IAR_PATH). It was not even enough to update PATH in local DOS prompt. I finally managed to fix it by setting Windows 10 System wide PATH and/or by hacking full path in the exported Makefile.

So I guess that "make error 2" should probably go to mbed CLI? Maybe you want to open a ticket for them :)

SeppoTakalo, 2017-06-14:
OK.
Then I will close this.
CLI related issues should be related to mbed-cli repo.

brandonboesch, 2017-06-14:
It has the path variable added for gcc-6, and is compiling all the c files, but the error happens at the end of the compile during the link process.

brandonboesch, 2017-06-14:
In regards to make_iar, I updated the path variable in windows as recommended, and tried the memory optimization recommended in the readme by adding the following to the mbed_app.json file:
"macros": ["MBEDTLS_USER_CONFIG_FILE=\"mbedtls_config.h\""]

It is now throwing the following errors:
[image]

theotherjimmy, 2017-06-14:
@brandonboesch Did you export after doing the memory optimization from the readme? Order matters!

brandonboesch, 2017-06-15:
@theotherjimmy, Yes, I added the memory optimization to mbed_app.json and then exported, in that order.

theotherjimmy, 2017-06-15:
@brandonboesch @SeppoTakalo Considering that this example has trouble fitting, could we resolve this by suggesting that users export with the "develop" or "release" profiles?

MarceloSalazar, 2017-06-27:
@brandonboesch @SeppoTakalo could you share your thoughts on @theotherjimmy question?

SeppoTakalo, 2017-09-18:
I have no knowledge how to export with "release" profile.
Can someone specify which parameter it is, help does not show it:
```
$ mbed export --help
usage: mbed export [-h] [-i IDE] [-m TARGET] [--source SOURCE] [-c] [-S] [-v]
                   [-vv]

Generate IDE project files for the current program.

optional arguments:
  -h, --help            show this help message and exit
  -i IDE, --ide IDE     IDE to create project files for. Example: UVISION4,
                        UVISION5, GCC_ARM, IAR, COIDE
  -m TARGET, --target TARGET
                        Export for target MCU. Example: K64F, NUCLEO_F401RE,
                        NRF51822...
  --source SOURCE       Source directory. Default: . (current dir)
  -c, --clean           Clean the build directory before compiling
  -S, --supported       Shows supported matrix of targets and toolchains
  -v, --verbose         Verbose diagnostic output
  -vv, --very_verbose   Very verbose diagnostic output
```

Otherwise, this whole issue is now memory-fitting problem for some platform.

MarceloSalazar, 2017-09-18:
https://docs.mbed.com/docs/mbed-os-handbook/en/latest/dev_tools/build_profiles/

SeppoTakalo, 2017-09-18:
I cannot test this. exporters seems to be failing https://github.com/ARMmbed/mbed-os/issues/5131

SeppoTakalo, 2017-09-18:
This issue will be fixed by appending the documentation.

PR #125 (2017-09-18): Add more memory optimization info.
Fixes #95
commit 226e22d7 (2017-09-18): Add more memory optimization info.

Fixes #95
closed 2017-09-18 by SeppoTakalo