ARMmbed/mbed-os-example-mesh-minimal #96: Application failures

AriParkkila, 2017-06-13:
Logging failures when application was build:
mbed compile -m NUCLEO_F411RE -t GCC_ARM --app-config configs/mesh_6lowpan.json

Log "pins not configured correctly" comes every time.

I restarted mesh-node immediately when border-router obtained IPv6 address. It seems that the node is working fine although UART log reports failure:
Connecting...
Connection failed! -3010
(there is nothing in the log after the line above)

SeppoTakalo, 2017-06-13:
The `Connection failed! -3010` is timeout failure, it did not find the network.
Are you sure that the node is in the same channel than the BR?

AriParkkila, 2017-06-14:
This issue is about logging and not application failure. I don't know this for sure but it seems that the mesh-node works fine after all - although log print without debug traces has only "Connection failed! -3010".

See log files attached. It seems that call to getOwnIpAddress() and thus print "connected. IP = 2001:999:3:8095:fec2:3d00:3:be00" is missing when first bootstrap failed.

[Log with connected print.](https://github.com/ARMmbed/mbed-os-example-mesh-minimal/files/1073656/success.txt)

[Log without connected print.](https://github.com/ARMmbed/mbed-os-example-mesh-minimal/files/1073657/fail.txt)

SeppoTakalo, 2017-06-14:
Closing as the device eventually joins the network and the error message is now fixed.

For joining part, there will be change coming in internal ticket ONME-3089 that will wait for as long as it requires for this device to join.

closed 2017-06-14 by SeppoTakalo