ARMmbed/mbed-os-example-mesh-minimal #97: Online compiler warnings

AriParkkila, 2017-06-13:
Warning: Transfer of control bypasses initialization of in "extras/mbed-os.lib/features/FEATURE_COMMON_PAL/nanostack-libservice/source/nsdynmemLIB/nsdynmemLIB.c", Line: 180, Col: 10
Warning: Transfer of control bypasses initialization of in "extras/mbed-os.lib/features/FEATURE_COMMON_PAL/nanostack-libservice/source/nsdynmemLIB/nsdynmemLIB.c", Line: 204, Col: 10
Warning: Variable "mutex" was declared but never referenced in "extras/mbed-os.lib/features/frameworks/greentea-client/source/greentea_metrics.cpp", Line: 37, Col: 29
Warning: Incompatible redefinition of macro "MBEDTLS_ERR_SSL_TIMEOUT" (declared at <a href="#" onmousedown="mbed_doc_goto('/mbed-os-example-mesh-minimal//extras/mbed-os.lib/features/mbedtls/inc/mbedtls/ssl.h', '107'); return false;">/extras/mbed-os.lib/features/mbedtls/inc/mbedtls/ssl.h:107</a>) in "extras/mbed-os.lib/features/nanostack/FEATURE_NANOSTACK/coap-service/source/include/coap_security_handler.h", Line: 103, Col: 10
Warning: Incompatible redefinition of macro "MBEDTLS_ERR_SSL_WANT_READ" (declared at <a href="#" onmousedown="mbed_doc_goto('/mbed-os-example-mesh-minimal//extras/mbed-os.lib/features/mbedtls/inc/mbedtls/ssl.h', '105'); return false;">/extras/mbed-os.lib/features/mbedtls/inc/mbedtls/ssl.h:105</a>) in "extras/mbed-os.lib/features/nanostack/FEATURE_NANOSTACK/coap-service/source/include/coap_security_handler.h", Line: 104, Col: 10
Warning: Incompatible redefinition of macro "MBEDTLS_ERR_SSL_WANT_WRITE" (declared at <a href="#" onmousedown="mbed_doc_goto('/mbed-os-example-mesh-minimal//extras/mbed-os.lib/features/mbedtls/inc/mbedtls/ssl.h', '106'); return false;">/extras/mbed-os.lib/features/mbedtls/inc/mbedtls/ssl.h:106</a>) in "extras/mbed-os.lib/features/nanostack/FEATURE_NANOSTACK/coap-service/source/include/coap_security_handler.h", Line: 105, Col: 10
Warning: Incompatible redefinition of macro "MBEDTLS_ERR_SSL_UNEXPECTED_MESSAGE" (declared at <a href="#" onmousedown="mbed_doc_goto('/mbed-os-example-mesh-minimal//extras/mbed-os.lib/features/mbedtls/inc/mbedtls/ssl.h', '74'); return false;">/extras/mbed-os.lib/features/mbedtls/inc/mbedtls/ssl.h:74</a>) in "extras/mbed-os.lib/features/nanostack/FEATURE_NANOSTACK/coap-service/source/include/coap_security_handler.h", Line: 106, Col: 10
Warning: Incompatible redefinition of macro "MBEDTLS_ERR_SSL_TIMEOUT" (declared at <a href="#" onmousedown="mbed_doc_goto('/mbed-os-example-mesh-minimal//extras/mbed-os.lib/features/mbedtls/inc/mbedtls/ssl.h', '107'); return false;">/extras/mbed-os.lib/features/mbedtls/inc/mbedtls/ssl.h:107</a>) in "extras/mbed-os.lib/features/nanostack/FEATURE_NANOSTACK/coap-service/source/include/coap_security_handler.h", Line: 103, Col: 10
Warning: Incompatible redefinition of macro "MBEDTLS_ERR_SSL_WANT_READ" (declared at <a href="#" onmousedown="mbed_doc_goto('/mbed-os-example-mesh-minimal//extras/mbed-os.lib/features/mbedtls/inc/mbedtls/ssl.h', '105'); return false;">/extras/mbed-os.lib/features/mbedtls/inc/mbedtls/ssl.h:105</a>) in "extras/mbed-os.lib/features/nanostack/FEATURE_NANOSTACK/coap-service/source/include/coap_security_handler.h", Line: 104, Col: 10
Warning: Incompatible redefinition of macro "MBEDTLS_ERR_SSL_WANT_WRITE" (declared at <a href="#" onmousedown="mbed_doc_goto('/mbed-os-example-mesh-minimal//extras/mbed-os.lib/features/mbedtls/inc/mbedtls/ssl.h', '106'); return false;">/extras/mbed-os.lib/features/mbedtls/inc/mbedtls/ssl.h:106</a>) in "extras/mbed-os.lib/features/nanostack/FEATURE_NANOSTACK/coap-service/source/include/coap_security_handler.h", Line: 105, Col: 10
Warning: Incompatible redefinition of macro "MBEDTLS_ERR_SSL_UNEXPECTED_MESSAGE" (declared at <a href="#" onmousedown="mbed_doc_goto('/mbed-os-example-mesh-minimal//extras/mbed-os.lib/features/mbedtls/inc/mbedtls/ssl.h', '74'); return false;">/extras/mbed-os.lib/features/mbedtls/inc/mbedtls/ssl.h:74</a>) in "extras/mbed-os.lib/features/nanostack/FEATURE_NANOSTACK/coap-service/source/include/coap_security_handler.h", Line: 106, Col: 10
Warning: Phy_register inherits implicit virtual in "extras/mbed-os.lib/features/nanostack/FEATURE_NANOSTACK/nanostack-interface/NanostackRfPhy.h", Line: 30, Col: 13
Warning: Connect inherits implicit virtual in "extras/mbed-os.lib/features/nanostack/FEATURE_NANOSTACK/mbed-mesh-api/mbed-mesh-api/LoWPANNDInterface.h", Line: 37, Col: 10
Warning: Disconnect inherits implicit virtual in "extras/mbed-os.lib/features/nanostack/FEATURE_NANOSTACK/mbed-mesh-api/mbed-mesh-api/LoWPANNDInterface.h", Line: 38, Col: 10
Warning: GetOwnIpAddress inherits implicit virtual in "extras/mbed-os.lib/features/nanostack/FEATURE_NANOSTACK/mbed-mesh-api/mbed-mesh-api/LoWPANNDInterface.h", Line: 39, Col: 11
Warning: Function "MeshInterfaceNanostack::initialize(NanostackPhy *)" is hidden by "LoWPANNDInterface::initialize" -- virtual function override intended? in "extras/mbed-os.lib/features/nanostack/FEATURE_NANOSTACK/mbed-mesh-api/mbed-mesh-api/LoWPANNDInterface.h", Line: 36, Col: 20
Warning: Connect inherits implicit virtual in "extras/mbed-os.lib/features/nanostack/FEATURE_NANOSTACK/mbed-mesh-api/mbed-mesh-api/ThreadInterface.h", Line: 37, Col: 10
Warning: Disconnect inherits implicit virtual in "extras/mbed-os.lib/features/nanostack/FEATURE_NANOSTACK/mbed-mesh-api/mbed-mesh-api/ThreadInterface.h", Line: 38, Col: 10
Warning: GetOwnIpAddress inherits implicit virtual in "extras/mbed-os.lib/features/nanostack/FEATURE_NANOSTACK/mbed-mesh-api/mbed-mesh-api/ThreadInterface.h", Line: 72, Col: 11
Warning: Function "MeshInterfaceNanostack::initialize(NanostackPhy *)" is hidden by "ThreadInterface::initialize" -- virtual function override intended? in "extras/mbed-os.lib/features/nanostack/FEATURE_NANOSTACK/mbed-mesh-api/mbed-mesh-api/ThreadInterface.h", Line: 36, Col: 20
Warning: Connect inherits implicit virtual in "extras/mbed-os.lib/features/nanostack/FEATURE_NANOSTACK/mbed-mesh-api/mbed-mesh-api/NanostackEthernetInterface.h", Line: 30, Col: 10
Warning: Disconnect inherits implicit virtual in "extras/mbed-os.lib/features/nanostack/FEATURE_NANOSTACK/mbed-mesh-api/mbed-mesh-api/NanostackEthernetInterface.h", Line: 31, Col: 10
Warning: GetOwnIpAddress inherits implicit virtual in "extras/mbed-os.lib/features/nanostack/FEATURE_NANOSTACK/mbed-mesh-api/mbed-mesh-api/NanostackEthernetInterface.h", Line: 32, Col: 11
Warning: Function "MeshInterfaceNanostack::initialize(NanostackPhy *)" is hidden by "NanostackEthernetInterface::initialize" -- virtual function override intended? in "extras/mbed-os.lib/features/nanostack/FEATURE_NANOSTACK/mbed-mesh-api/mbed-mesh-api/NanostackEthernetInterface.h", Line: 29, Col: 20
Warning: Phy_register inherits implicit virtual in "extras/mbed-os.lib/features/nanostack/FEATURE_NANOSTACK/nanostack-interface/NanostackRfPhy.h", Line: 30, Col: 13
Warning: Connect inherits implicit virtual in "extras/mbed-os.lib/features/nanostack/FEATURE_NANOSTACK/mbed-mesh-api/mbed-mesh-api/LoWPANNDInterface.h", Line: 37, Col: 10
Warning: Disconnect inherits implicit virtual in "extras/mbed-os.lib/features/nanostack/FEATURE_NANOSTACK/mbed-mesh-api/mbed-mesh-api/LoWPANNDInterface.h", Line: 38, Col: 10
Warning: GetOwnIpAddress inherits implicit virtual in "extras/mbed-os.lib/features/nanostack/FEATURE_NANOSTACK/mbed-mesh-api/mbed-mesh-api/LoWPANNDInterface.h", Line: 39, Col: 11
Warning: Function "MeshInterfaceNanostack::initialize(NanostackPhy *)" is hidden by "LoWPANNDInterface::initialize" -- virtual function override intended? in "extras/mbed-os.lib/features/nanostack/FEATURE_NANOSTACK/mbed-mesh-api/mbed-mesh-api/LoWPANNDInterface.h", Line: 36, Col: 20
Warning: Connect inherits implicit virtual in "extras/mbed-os.lib/features/nanostack/FEATURE_NANOSTACK/mbed-mesh-api/mbed-mesh-api/ThreadInterface.h", Line: 37, Col: 10
Warning: Disconnect inherits implicit virtual in "extras/mbed-os.lib/features/nanostack/FEATURE_NANOSTACK/mbed-mesh-api/mbed-mesh-api/ThreadInterface.h", Line: 38, Col: 10
Warning: GetOwnIpAddress inherits implicit virtual in "extras/mbed-os.lib/features/nanostack/FEATURE_NANOSTACK/mbed-mesh-api/mbed-mesh-api/ThreadInterface.h", Line: 72, Col: 11
Warning: Function "MeshInterfaceNanostack::initialize(NanostackPhy *)" is hidden by "ThreadInterface::initialize" -- virtual function override intended? in "extras/mbed-os.lib/features/nanostack/FEATURE_NANOSTACK/mbed-mesh-api/mbed-mesh-api/ThreadInterface.h", Line: 36, Col: 20
[... cut ...]
Warning: Set_mac_address inherits implicit virtual in "atmel-rf-driver/atmel-rf-driver/NanostackRfPhyAtmel.h", Line: 64, Col: 11
Warning: #warning directive: toolchain.h has been replaced by mbed_toolchain.h, please update to mbed_toolchain.h [since mbed-os-5.3] in "extras/mbed-os.lib/platform/toolchain.h", Line: 23, Col: 3
Success: Success! in "<a href="#" onClick="mbed_navigate_to('/mbed-os-example-mesh-minimal', function() { theitf.controls.get('sidebar_tabs').activate(1); }); return false;">Build Details</a>"

SeppoTakalo, 2017-06-13:
Can you specify the toolchain and target here. Thanks.

AriParkkila, 2017-06-13:
How can I see which toolchain online compiler uses?

SeppoTakalo, 2017-06-13:
Ah, sorry. I did not spot the word `online` from the title.
That answers the question.

SeppoTakalo, 2017-06-14:
Closing as these are duplicate of the https://github.com/ARMmbed/mbed-os-example-mesh-minimal/issues/90

Same warnings are visible in the ARM build.

closed 2017-06-14 by SeppoTakalo