ARMmbed/mbed-os-example-mesh-minimal #100: led control example not functional on NCS36510 in various configurations of Thread devices

maclobdell, 2017-06-14:
Testing version 5.5 release candidate 2.

toolchains tested: IAR

I used all the various methods to reduce the memory footprint that are listed in the readme, I also compiled with the release profile (which I recommend be added to the list also!)

`
mbed compile --profile mbed-os/tools/profiles/release.json
`

It is able to connect to the border router and get an IP address, but it does not turn on the led when you press the button on another board running the same application. 

I tried Thread sleepy end node, minimal end node, and router configurations.  Each seem to have their own problem.  

I suspect that the sleepy end node does not listen for multicast messages, and the router node runs out of memory.  

not sure why the minimal end node does not work, because I thought that was added to address this problem.  
cc @karsev
 
Here are the various mbed_app.json settings I tried.

Sleepy end device

```
{
    "config": {
        "radio-type":{
            "help": "options are ATMEL, MCR20, NCS36510, KW24D",
            "value": "NCS36510"
        },
        "mesh-type":{
            "help": "options are MESH_LOWPAN, MESH_THREAD",
            "value": "MESH_THREAD"
        },
        "enable-led-control-example": true,
        "LED": "NC",
        "BUTTON": "NC"
    },
    "macros": ["MBEDTLS_USER_CONFIG_FILE=\"mbedtls_config.h\""],
    "target_overrides": {
        "*": {
            "target.features_add": ["NANOSTACK", "THREAD_END_DEVICE", "COMMON_PAL"],
            "mbed-trace.enable": true,
            "nanostack-hal.event_loop_thread_stack_size": 2048,
            "mbed-mesh-api.heap-size": 14000,
            "mbed-mesh-api.thread-device-type": "MESH_DEVICE_TYPE_THREAD_SLEEPY_END_DEVICE",
            "platform.stdio-convert-newlines": true,
            "platform.stdio-baud-rate": 115200
        },
        "NCS36510": {
            "LED": "LED1",
            "BUTTON": "SW1"
        }
    }
}
```

Minimal end device

```
{
    "config": {
        "radio-type":{
            "help": "options are ATMEL, MCR20, NCS36510, KW24D",
            "value": "NCS36510"
        },
        "mesh-type":{
            "help": "options are MESH_LOWPAN, MESH_THREAD",
            "value": "MESH_THREAD"
        },
        "enable-led-control-example": true,
        "LED": "NC",
        "BUTTON": "NC"
    },
    "macros": ["MBEDTLS_USER_CONFIG_FILE=\"mbedtls_config.h\""],
    "target_overrides": {
        "*": {
            "target.features_add": ["NANOSTACK", "THREAD_ROUTER", "COMMON_PAL"],
            "mbed-trace.enable": true,
            "nanostack-hal.event_loop_thread_stack_size": 2048,
            "mbed-mesh-api.heap-size": 14000,
            "mbed-mesh-api.thread-device-type": "MESH_DEVICE_TYPE_THREAD_MINIMAL_END_DEVICE",
            "platform.stdio-convert-newlines": true,
            "platform.stdio-baud-rate": 115200
        },
        "NCS36510": {
            "LED": "LED1",
            "BUTTON": "SW1"
        }
    }
}
```

Router 

```
{
    "config": {
        "radio-type":{
            "help": "options are ATMEL, MCR20, NCS36510, KW24D",
            "value": "NCS36510"
        },
        "mesh-type":{
            "help": "options are MESH_LOWPAN, MESH_THREAD",
            "value": "MESH_THREAD"
        },
        "enable-led-control-example": true,
        "LED": "NC",
        "BUTTON": "NC"
    },
    "macros": ["MBEDTLS_USER_CONFIG_FILE=\"mbedtls_config.h\""],
    "target_overrides": {
        "*": {
            "target.features_add": ["NANOSTACK", "THREAD_ROUTER", "COMMON_PAL"],
            "mbed-trace.enable": true,
            "nanostack-hal.event_loop_thread_stack_size": 2048,
            "mbed-mesh-api.heap-size": 14000,
            "mbed-mesh-api.thread-device-type": "MESH_DEVICE_TYPE_THREAD_ROUTER",
            "platform.stdio-convert-newlines": true,
            "platform.stdio-baud-rate": 115200
        },
        "NCS36510": {
            "LED": "LED1",
            "BUTTON": "SW1"
        }
    }
}
```

By the way, how about including an example config file for each of these configurations with the example?

karsev, 2017-06-14:
The example works with Thread router and MED devices. 
When you define MED device you should define THREAD_END_DEVICE -> 
"target.features_add": ["NANOSTACK", "THREAD_END_DEVICE", "COMMON_PAL"]

I tried out the example on NCS36510 and it seems to be very unstable. Lot of messages are send and received and after some time device runs out of memory. Also the button does not create ISR events always.  So the issue seems to be that there is not enough memory for the application.

SeppoTakalo, 2017-06-14:
@artokin @mikter @karsev Can you investigate?

karsev, 2017-06-14:
I found that the button gpio is floating by default. It can be fixed in -> targets/TARGET_ONSEMI/TARGET_NCS36510/PinNames.h
by defining in the PinMode enum -> 
PullDefault       = PushPullPullUp.

To make sure there is enough RAM I did the following ->
In Rtos/mbed_boot.c -> MBED_CONF_APP_MAIN_STACK_SIZE -> changed to 2kb
Rtos/rtx5/Mbed_rtx_conf.h -> MBED_CONF_APP_THREAD_STACK_SIZE -> changed to 1kb

With the GCC compiler seemed to work.
Still sometimes packet does not get delivered to other nodes.

Another issue was with the ARM compiler reporting wrong RAM numbers when using mbed compile.

maclobdell, 2017-06-15:
Thanks a lot Kari.  I think that the internal pullup should be enabled for just the push button switch, but not other pins because that will have an impact on power consumption and might cause other problems.   

I think this can be fixed by setting the pullup in mesh_led_control_example.cpp.  This should work with boards that ground the signal when the button is pressed.  I expect that to be the vast majority of boards.  Maybe a new configuration option can be added in mbed_app.json to select whether to enable a pullup or not.  

    if (MBED_CONF_APP_BUTTON != NC) {
    	my_button.fall(&my_button_isr);
        my_button.mode(PullUp);
    }

For the main stack and thread stack changes, can this be done in the mbed_app.json file correct?  Can you clarify?  Can this example include an additional config file for reference in the configs folder that includes all the various optimizations for a small memory device?

karsev, 2017-06-16:
I have added the pullUp mode definition in the led example.
The example works without any rtx stack size changes, as MED mode does not require some much memory.

MarceloSalazar, 2017-06-27:
@maclobdell @karsev is this issue resolved (link to PR needed) or requires a fix?

maclobdell, 2017-06-28:
fixed by https://github.com/ARMmbed/mbed-os-example-mesh-minimal/pull/103.

closed 2017-06-28 by maclobdell