ARMmbed/mbed-os-example-mesh-minimal #117: Updates not getting pushed to developer.mbed.org

maclobdell, 2017-08-03:
Many of the mbed os examples from the ARMmbed github team are mirrored on the developer.mbed.org website.  If I am correct, each mbed OS release cycle triggers updates to the official examples, and pushes the updates to the special mbed-os-examples team on mbed.  For example, mbed os just had a patch release and the blinky example was updated and pushed to https://developer.mbed.org/teams/mbed-os-examples/code/mbed-os-example-blinky/

Looks like the mechanism for pushing update for the mesh minimal repo on developer.mbed.org is not working.  It is 4 months old, whereas the last release on github was under 1 month ago.
https://developer.mbed.org/teams/mbed-os-examples/code/mbed-os-example-mesh-minimal/

This is very useful because it allows examples to be associated with specific platforms as templates for the platform.  (e.g. FRDM-KW24D)

SeppoTakalo, 2017-09-13:
@maclobdell @MarceloSalazar @adbridge Do you know who have maintained the cloning job?
I don't believe that I have cloned this manually. There was some kind of automation at some point.
I have no idea how to help.

MarceloSalazar, 2017-09-13:
@maclobdell thanks for bringing this topic, indeed the repos are out of sync.

@screamerbg as we looked into this some time ago, is this something you or someone else could review and fix?

SeppoTakalo, 2017-09-20:
ping @adbridge and @screamerbg Do you have any knowledge about this cloning? Can you help us resolve this?

adbridge, 2017-09-20:
@SeppoTakalo unfortunately I don't know how this gets cloned over to mbed org either. 
@theotherjimmy any idea?

screamerbg, 2017-09-20:
@SeppoTakalo @adbridge the minimal-mesh repository has been rewritten. The mirroring process is very sensitive about repo rewrites and stops mirroring if rewrite is detected. Will fix this shortly.

adbridge, 2017-09-20:
@SeppoTakalo Mihail has fixed the script, please check that the repo is mirrored again correctly.

SeppoTakalo, 2017-09-26:
[image]
It is cloning now. Thanks for fixing this.

closed 2017-09-26 by SeppoTakalo