ARMmbed/mbed-os-example-mesh-minimal #121: Overcommitted RAM on NCS36510 w/ IAR

jacobjohnson-ON, 2017-08-31:
I am trying to get the latest version this demo to run on the NCS36510, which has 48k of RAM.  When I use [this mbed_app.json file](https://github.com/ARMmbed/mbed-os-example-mesh-minimal/files/1268626/mbed_app.txt) it compiles all right.  Notice the extreme memory optimizations.  From my experience with this example app, it would not run well if at all with this configuration.  
[Here is the output on the CLI](https://github.com/ARMmbed/mbed-os-example-mesh-minimal/files/1268629/CLI.txt) and 
[the linker map file int the BUILD folder](https://github.com/ARMmbed/mbed-os-example-mesh-minimal/files/1268630/mbed-os-example-mesh-minimal.txt).  The CLI reports 28k RAM usage, but the linker map reports 45k RAM.  Not sure which one is the right one.  

When I increase the mbed-mesh-api heap size by 1 kB, and the event_loop_thread_stack_size by 1kB (Now at the recommended max optimization for end device configuration), the linker fails, claiming the RAM memory is now overcommitted.  [(See the CLI output here)](https://github.com/ARMmbed/mbed-os-example-mesh-minimal/files/1268648/CLI_Fail.txt), [the .json file](https://github.com/ARMmbed/mbed-os-example-mesh-minimal/files/1268650/mbed_app_Fail.txt), [and the linker map](https://github.com/ARMmbed/mbed-os-example-mesh-minimal/files/1268651/mbed-os-example-mesh-minimal_Fail.txt)

When I compile this app with the same configuration using the GCC compiler, it compiles and links just fine.  It runs the LED control example all right, though after about a hundred messages the device begins to report memory failures on the terminal window, and the messages no longer are received by the other devices in the network.

maclobdell, 2017-09-01:
I did some experiments and found that reducing the heap allocated in the IAR linker file NCS36510.icf can get it to fit.  However, I have not run the full mbed-os test suite to see if there are any problems with this change.  

- define symbol __ICFEDIT_size_heap__   = 0x4000;
+ define symbol __ICFEDIT_size_heap__   = 0x3000;

Note that IAR requires that the heap size be explicitly set where the other compilers allow the size to be automatically generated using all the remaining available ram.  

The ram usage reported by mbed cli should include both statically allocated data and dynamically allocated heap/stack.  There might be an issue where CLI can't detect the dynamic memory size on this platform, so it is not showing it.  I think that is where the difference between 28K and 45K.

danclement, 2017-09-01:
@maclobdell, how would we know how big the heap needs to be?  I'm sure it's quite different if running a Thread application versus a generic mbed OS application.  Is there a best practices guide?  Or is it too target specific?  Just wondering so we can either bake it in somehow so it's automatic, or in the worst case to make a wiki page to explain what to do and how.  Thanks.

jacobjohnson-ON, 2017-09-01:
@maclobdell Thanks for the suggestion.  I made the change and was able to get the files to link and run on the device, however there is still a memory issue that causes the Demo to stop working after around 50 messages are sent back and forth.  The mbed trace indicates that it is a memory problem.  Here is the trace output when the connection broke: 

[DBG ][nsif]: socket_recvfrom(socket=3fff8ed4) sock_id=3, ret=-3001, addr=[fd00:db8::ff:fe00:9c02]:1234
[DBG ][example]: send msg 0
[DBG ][nsif]: socket_sendto(socket=3fff8ed4) sock_id=3, ret=3
[DBG ][nsif]: socket_callback() sock=3, event=0, interface=1, data len=3
[DBG ][nsif]: SOCKET_DATA, sock=3, bytes=3
[DBG ][nsif]: socket_recvfrom(socket=3fff8ed4) sock_id=3, ret=3, addr=[fd00:db8::ff:fe00:9c02]:1234
[DBG ][example]: Packet from fd00:0db8:0000:0000:0000:00ff:fe00:9c02

[DBG ][example]: Advertisiment after 39 seconds
[DBG ][example]: Turning led off

[DBG ][nsif]: socket_recvfrom(socket=3fff8ed4) sock_id=3, ret=-3001, addr=[fd00:db8::ff:fe00:9c02]:1234
[DBG ][example]: send msg 1
[DBG ][nsif]: socket_sendto(socket=3fff8ed4) sock_id=3, ret=2
[DBG ][example]: send msg 0
[ERR ][nsif]: socket_sendmsg: error=-2
[DBG ][nsif]: socket_sendto(socket=3fff8ed4) sock_id=3, ret=-3012
[DBG ][example]: send msg 1
[ERR ][nsif]: socket_sendmsg: error=-2
[DBG ][nsif]: socket_sendto(socket=3fff8ed4) sock_id=3, ret=-3012
[DBG ][example]: send msg 0
[ERR ][nsif]: socket_sendmsg: error=-2
[DBG ][nsif]: socket_sendto(socket=3fff8ed4) sock_id=3, ret=-3012
[DBG ][example]: send msg 1
[ERR ][nsif]: socket_sendmsg: error=-2
[DBG ][nsif]: socket_sendto(socket=3fff8ed4) sock_id=3, ret=-3012
[DBG ][example]: send msg 0
[ERR ][nsif]: socket_sendmsg: error=-2
[DBG ][nsif]: socket_sendto(socket=3fff8ed4) sock_id=3, ret=-3012

cc @danclement

maclobdell, 2017-09-05:
@karsev - Can you review this issue?  Is there memory optimization that can be done to the demo application?  For this simple application that is sending/receiving multicast packets, can it clean up and reallocate memory to avoid running out of heap?  Can you provide recommendations on testing / debugging this issue to determine where the memory is being used up?

karsev, 2017-09-06:
There are some options in the mbed-os. Here they are:
- In arm_hal_timer.cpp (mbed-os\features\FEATURE_COMMON_PAL\nanostack-hal-mbed-cmsis-rtos)
   - static uint64_t timer_thread_stk[2048/sizeof(uint64_t)]; (can be set to 1024)
- In Mbed_rtx_conf.h (mbed-os/rtos/rtx5)
   - #define MBED_CONF_APP_THREAD_STACK_SIZE 4096 (2048 should be fine)
   - #define OS_TIMER_THREAD_STACK_SIZE 768 (can be set to zero, if RTOSTimer is not used)
- In mbed_boot.c (mbed-os/rtos)
   - #define MBED_CONF_APP_MAIN_STACK_SIZE 4096 (can be set to 2048)

As there should be now more free RAM, you could check if these definitions could be set a little bit higher.
- "mbed-mesh-api.heap-size": 13000, (14-15kb)
- "nanostack-hal.event_loop_thread_stack_size": 1048 (2kb)

jacobjohnson-ON, 2017-09-07:
Making the suggested changes, I was able to increase the mbed-mesh-api.heap-size to 16k and the event_loop_thread_stack_size to 2 kb.  This made the problem better, however the messages eventually stopped being communicated properly.  The communication broke especially quickly when I sent alternating messages from the two end nodes.  

For comparison, I loaded the binary of this application that I compiled back in June, using the GCC compiler, and I was able to run the LED demo with no problems at all.

danclement, 2017-09-07:
@jacobjohnson-ON, would it make sense to you to try running a software controlled transmission/reception of messages so that the messages are occuring in a repeatable and predictable way?  I have the feeling this might be a memory leak and running a test like this in a controlled way we may be able to isolate if there is a memory leak or some other predictable issue by knowing exactly when it crashes.

My preference would be for ARM to design and run this test as it looks like an issue in their stack, or in the port to our device more than a memory limitation in hardware.  @maclobdell what do you think?

jacobjohnson-ON, 2017-09-07:
That would certainly be useful.

danclement, 2017-09-07:
@maclobdell please outline the formal request for ON Semiconductor to request Thread support for our device.  Thanks!

SeppoTakalo, 2017-09-18:
We have recorded the memory optimisations as internal requirements for the OS. In the mean time, there is not much to do within the scope of this application.

IAR version 8 might help as to my understanding it brings in dynamic heap support. We are working towards supporting IAR 8.1 in Mbed OS.

Closing is as the work to support this combination will not be done within this application.
Thanks for the feedback.

closed 2017-09-18 by SeppoTakalo