ARMmbed/mbed-os-example-mesh-minimal #122: mbed-os-example-mesh-minimal is failing with radio-type set to ATMEL on K64F

SenRamakri, 2017-09-01:
Hello,

I'm trying to run mbed-os-example-mesh-minimal on K64F with ATMEL sheild(2.4) and its failing as below.

Connecting...

Connection failed! -3012

The mesh-type is set to MESH_THREAD. The code runs successfully with MCR20 though, and generates following output. 
Connecting...

connected. IP = fd00:db80::8c19:9b4a:fe08:bf74

SeppoTakalo, 2017-09-18:
Hi @SenRamakri 

What kind of test environment are you running?
Can you share the `mbed_app.json` files from both border router and mesh client application.

Thanks.

SeppoTakalo, 2017-10-04:
Closing because not enough information is provided.
Both of these configurations are part of our CI testing and are working fine.
I cannot see any problems here.

Please re-open if you have more information or the problem is still valid.

closed 2017-10-04 by SeppoTakalo