ARMmbed/mbed-os-example-mesh-minimal #140: Example fails with Mutex error

balamk07, 2017-10-30:
Hi,

The example fails with below error message after few seconds of execution

_Mutex 0x200011fc error -6: Not allowed in ISR context_

Example built with mbed-os 5.6.1. My test setup has three K64F + Atmel radio board, one configured as router and other two as SED. I was able to get the example to work reliably only with rev 5.4.7 mbeb-os.

Thanks,
Bala.

SeppoTakalo, 2017-10-31:
Hi @balamk07 

When you update the Mbed OS 5.6, can you try `mbed update HEAD` from the root folder.

This sounds like a driver issue that has long time since been fixed. Maybe they are outdated.

How did you actually update to 5.6?

Can you replicate the effect if you do fresh clone:
```
mbed import mbed-os-example-mesh-minimal
cd mbed-os-example-mesh-minimal
mbed compile ...
```

If the problem persist, please try attaching a debugger (for example pyOCD) and provide us a stack trace.

balamk07, 2017-11-01:
Hi @SeppoTakalo 

I have been using only debug build of this example all this time. The example only fails in debug build, release build works fine with mbed os 5.6.1. Could you please try the example in debug build? I hope you could replicate the issue.

Thanks,
Bala.

SeppoTakalo, 2017-11-01:
It works because release build does not assert().

Did you try to replicate this with clean, up-to-date, clone of this application?

balamk07, 2017-11-13:
The mutex error still occur with the clean clone with up-to-date mbed-os. Please refer to attached image. I can use release build without issues as there is no assert and it keeps running. debug build crashes with this error. 

The error appears when:
- SW2 on SED device gets pressed, which turns on router LED
- Takes about 50 seconds after button press for mutes fail error

[image]

SeppoTakalo, 2017-11-30:
Fixed in https://github.com/ARMmbed/mbed-os-example-mesh-minimal/pull/148

closed 2017-11-30 by SeppoTakalo