ARMmbed/mbed-os-example-mesh-minimal #152: Conflicting App Config in Documentation

sarahmarshy, 2017-12-08:
Told [here](https://github.com/ARMmbed/mbed-os-example-mesh-minimal#selecting-optimal-nanostack-configuration) to modify `mbed_app.json`

Told to [compile](https://github.com/ARMmbed/mbed-os-example-mesh-minimal#compile-the-application) with `configs/mesh_6lowpan.json`. `mesh_6lowpan.json` not mentioned or explained previously.

MarceloSalazar, 2017-12-11:
I agree with @sarahmarshy , these are confusing steps.
We shouldn't be asking the user to edit json files.

@SeppoTakalo can we remove instructions to edit json file **and** also delete the `mbed_app.json` in the root directory? This remove conflicting messages.

SeppoTakalo, 2017-12-15:
Fixed in #162