ARMmbed/mbed-os-example-mesh-minimal #155: NUCLEO F401RE + MCR20 fails to compile for IAR

sarahmarshy, 2017-12-08:
Compilation error:
```
Compile [ 43.3%]: coap_security_handler.c
[Error] ssl.h@384,0: [Pe169]: expected a declaration
[ERROR]
  };
  ^
"C:\Repos\mbed-os-example-mesh-minimal\mbed-os\features\mbedtls\inc\mbedtls\ssl.h",384  Error[Pe169]: expected a declaration

[mbed] ERROR: "c:\python27\python.exe" returned error code 1.
[mbed] ERROR: Command "c:\python27\python.exe -u C:\Repos\mbed-os-example-mesh-minimal\mbed-os\tools\make.py -t iar -m NUCLEO_F401RE --source . --build .\BUILD\NUCLEO_F401RE\iar -c" in "C:\Repos\mbed-os-example-mesh-minimal"
---
```

MarceloSalazar, 2017-12-11:
@sarahmarshy can you please confirm you're using the OOB 5.7 branch?

FYI @SeppoTakalo are you aware of this?

artokin, 2017-12-11:
PR in review https://github.com/ARMmbed/coap-service/pull/86

sarahmarshy, 2017-12-11:
@MarceloSalazar, yes I can confirm.

SeppoTakalo, 2017-12-11:
@MarceloSalazar Yes, I'm now aware of this. Sound like we had this issue long way back with no fix.

Problem is somewhere in CoAP service or Mbed TLS. IAR is not liking SSL structures to be empty. I don't know why.
Also, I thought that this was already fixed https://github.com/ARMmbed/mbed-os-example-mesh-minimal/issues/91
Obviously the IAR build has been skipped for F401RE since.

SeppoTakalo, 2017-12-13:
Fixes merged to master.
Needs re-testing in RC2

MarceloSalazar, 2017-12-18:
Fixed confirmed by @sarahmarshy

closed 2017-12-18 by MarceloSalazar