ARMmbed/mbed-os-example-mesh-minimal #156: Links in documentation are obsolete

MarceloSalazar, 2017-12-11:
Having reviewed the [5.7 branch](https://github.com/ARMmbed/mbed-os-example-mesh-minimal/tree/mbed-os-5.7.0-oob) (before release), I find many links are obsolete or broken.

- 6LoWPAN overview , [link](https://os.mbed.com/docs/v5.6/tutorials/6lowpan-mesh.html) broken
- mbed Mesh API , shouldn't be [this](https://os.mbed.com/docs/latest/reference/mesh.html)  instead of [this](https://github.com/ARMmbed/mbed-os/blob/master/features/nanostack/FEATURE_NANOSTACK/mbed-mesh-api/README.md) one?
- TLS Porting guide. It seems [this](https://docs.mbed.com/docs/mbed-os-handbook/en/5.2/advanced/tls_porting/) is old. Where is the latest?
- The [configuration](https://github.com/ARMmbed/mbed-os/blob/master/features/nanostack/FEATURE_NANOSTACK/mbed-mesh-api/README.md#configurable-parameters-in-section-mbed-mesh-api) options seem very useful. Can we have this in os.mbed.com/docs ?
- The [profiles](https://docs.mbed.com/docs/mbed-os-handbook/en/latest/dev_tools/build_profiles/) is very useful, where can we find this in latest docs?

@AnotherButler can you please help us to identify links and docs? Thansk!

AnotherButler, 2017-12-11:
I've updated some of these on the OOB spreadsheet.

- 6LoWPAN overview: https://os.mbed.com/docs/v5.6/tutorials/mesh.html
- Yes, it should.
- That is the latest one I know of. TLS is getting its own microsite, so we did not migrate it to os.mbed.com/docs. However, I don't think that page is up on the TLS site yet.
- The [mesh tutorial](https://os-doc-builder.test.mbed.com/docs/v5.7/tutorials/mesh.html) does discuss configurable parameters. We also have a [connectivity configuration page](https://os.mbed.com/docs/v5.6/reference/configuration-connectivity.html), but it does not yet have the mesh content. I believe there's a JIRA ticket for that.
- Build profiles: https://os.mbed.com/docs/v5.6/tools/build-profiles.html

MarceloSalazar, 2017-12-12:
Thanks @AnotherButler ! A few notes for your reference.

- The [mesh configuration](https://github.com/ARMmbed/mbed-os/blob/master/features/nanostack/FEATURE_NANOSTACK/mbed-mesh-api/README.md#configurable-parameters-in-section-mbed-mesh-api) is only available on github, but not in the [tutorial](https://os-doc-builder.test.mbed.com/docs/v5.7/tutorials/mesh.html). Would be nice to bring this to docs in the future.
- I personally don't find the [connectivity](https://os.mbed.com/docs/v5.6/reference/configuration-connectivity.html) section useful, it's probably lacking more information and better formatting split per type of connectivity.

SeppoTakalo, 2017-12-13:
@MarceloSalazar @AnotherButler The Mesh API configuration values were part of OS documentation but I cannot find them anymore. Too much restructuring have happened. Or at least the text search does not help.

I'll try cloning the docs and then grepping through.

SeppoTakalo, 2017-12-13:
It seems to be in https://os.mbed.com/docs/v5.6/reference/network-socket.html
If you scroll down to bottom of the page.
Why its there, I don't know.

Now its mixed with Socket documentation and cellular modem documentation. I think this is unreadable.

@AnotherButler Can you move the "Module configuration" section under socket/mesh.

I'm closing this one as Marcelo already fixed the links.

closed 2017-12-13 by SeppoTakalo

AnotherButler, 2017-12-13:
@SeppoTakalo Can we move this content into the connectivity configuration section instead? We're trying to keep the configuration information there instead of in the API references.

SeppoTakalo, 2017-12-14:
It sounds like a correct place, but then again, those values do not make any sense for those who have not studied the Mesh chapters before.
So there is a danger that we make it sound like too complicated process to configure. In reality, defaults are fine for most of the users. Mesh networks require fine tuning.

AnotherButler, 2017-12-15:
Then let's move it to the configuration page for now and see what people think.