ARMmbed/mbed-os-example-mesh-minimal #175: Exporting to any Makefile based fails to build

SeppoTakalo, 2018-03-07:
I did a quick exporter test using K64F

|           Tool |Result |
|----------------|-------|
|       uvision5 |  PASS |
|            iar |  PASS |
|     make_armc5 |  FAIL |
|   make_gcc_arm |  FAIL |
|       make_iar |  FAIL |

So all Makefile based exporters fails to build.

Manually testing the `make_armc5` looks like the failure happens at the linkin phase
```
link: mbed-os-example-mesh-minimal.elf
process_begin: CreateProcess(C:\Keil_v5\ARM\ARMCC\bin\armlink.exe, armlink --show_full_path --scatter .././.link_script.sct --output mbed-os-example-mesh-minimal.elf atmel-rf-driver/source/NanostackRfPhyAtmel.o atmel-rf-driver/source/at24mac.o main.o mbed-os/drivers/AnalogIn.o mbed-os/drivers/BusIn.o mbed-os/drivers/BusInOut.o mbed-os/drivers/BusOut.o mbed-os/drivers/CAN.o mbed-os/drivers/Ethernet.o mbed-os/drivers/FlashIAP.o mbed-os/drivers/I2C.o mbed-os/drivers/I2CSlave.o mbed-os/drivers/InterruptIn.o mbed-os/drivers/InterruptManager.o mbed-os/drivers/MbedCRC.o mbed-os/drivers/RawSerial.o mbed-os/drivers/SPI.o mbed-os/drivers/SPISlave.o mbed-os/drivers/Serial.o mbed-os/drivers/SerialBase.o mbed-os/drivers/TableCRC.o mbed-os/drivers/Ticker.o mbed-os/drivers/Timeout.o mbed-os/drivers/Timer.o mbed-os/drivers/TimerEvent.o mbed-os/drivers/UARTSerial.o mbed-os/events/EventQueue.o mbed-os/events/equeue/equeue.o mbed-os/events/equeue/equeue_mbed.o mbed-os/events/equeue/equeue_posix.o mbed-os/events/mbed_shared_queues.o mbed-os/features/FEATURE_COMMON_PAL/mbed-client-randlib/source/randLIB.o mbed-os/features/FEATURE_COMMON_PAL/mbed-coap/source/sn_coap_builder.o mbed-os/features/FEATURE_COMMON_PAL/mbed-coap/source/sn_coap_header_check.o mbed-os/features/FEATURE_COMMON_PAL/mbed-coap/source/sn_coap_parser.o mbed-os/features/FEATURE_COMMON_PAL/mbed-coap/source/sn_coap_protocol.o mbed-os/features/FEATURE_COMMON_PAL/mbed-trace/source/mbed_trace.o mbed-os/features/FEATURE_COMMON_PAL/nanostack-hal-mbed-cmsis-rtos/arm_hal_interrupt.o mbed-os/features/FEATURE_COMMON_PAL/nanostack-hal-mbed-cmsis-rtos/arm_hal_random.o mbed-os/features/FEATURE_COMMON_PAL/nanostack-hal-mbed-cmsis-rtos/arm_hal_timer.o mbed-os/features/FEATURE_COMMON_PAL/nanostack-hal-mbed-cmsis-rtos/cs_nvm/cs_nvm.o mbed-os/features/FEATURE_COMMON_PAL/nanostack-hal-mbed-cmsis-rtos/ns_event_loop.o mbed-os/features/FEATURE_COMMON_PAL/nanostack-hal-mbed-cmsis-rtos/ns_hal_init.o mbed-os/features/FEATURE_COMMON_PAL/nanostack-hal-mbed-cmsis-rtos/nvm/nvm_ram.o mbed-os/features/FEATURE_COMMON_PAL/nanostack-libservice/source/IPv6_fcf_lib/ip_fsc.o mbed-os/features/FEATURE_COMMON_PAL/nanostack-libservice/source/libBits/common_functions
... clipped ...
make (e=87): The parameter is incorrect.
make[1]: *** [mbed-os-example-mesh-minimal.elf] Virhe 87
make: *** [all] Virhe 2

```

SeppoTakalo, 2018-03-07:
Googling the error code `make (e=87)` looks like this might be Windows issue
https://stackoverflow.com/questions/12598933/ndk-build-createprocess-make-e-87-the-parameter-is-incorrect

SeppoTakalo, 2018-03-07:
**Conclusion:** Windows CreateProcess() only accepts command line maximum length to be 32,768 characters.

In this case the final link command is 44,252 characters long.

So not an issue that could be fixed here.

adbridge, 2018-03-12:
@SeppoTakalo Looks like this should be moved to an mbed-os issue (against the tools) ?

SeppoTakalo, 2018-03-12:
Reported here: https://github.com/ARMmbed/mbed-os/issues/6335

closed 2018-03-12 by SeppoTakalo