ARMmbed/mbed-os-example-mesh-minimal #183: RTOS problem

drvrh, 2018-04-25:
When I run mbed-os-example-mesh-minimal with settings for NUCLEO_L152RE and changed config.json file. I get the following error and debugging mode. I attach a print screen output log after compiling.

MBED_APP.json

```
{
    "config": {
        "radio-type":{
            "help": "options are ATMEL, MCR20, NCS36510, KW24D",
            "value": "SPIRIT1"
        },
        "mesh-type":{
            "help": "options are MESH_LOWPAN, MESH_THREAD",
            "value": "MESH_LOWPAN"
        },
        "enable-led-control-example": true,
        "LED": "NC",
        "BUTTON": "NC"
    },
    "target_overrides": {
        "*": {
	    "spirit1.mac-address": "{0xf0, 0xf1, 0xf2, 0xf3, 0xf4, 0xf5, 0xf6, 0xf7}",
            "target.features_add": ["NANOSTACK", "COMMON_PAL"],
            "nanostack.configuration": "lowpan_host",
            "mbed-mesh-api.6lowpan-nd-panid-filter": "0xffff",
            "mbed-mesh-api.6lowpan-nd-channel-page": 0,
            "mbed-mesh-api.6lowpan-nd-channel": 12,
            "mbed-mesh-api.6lowpan-nd-channel-mask": "(1<<12)",
            "mbed-mesh-api.thread-config-panid": "0x0700",
            "mbed-mesh-api.thread-master-key": "{0x10, 0x11, 0x22, 0x33, 0x44, 0x55, 0x66, 0x77, 0x88, 0x99, 0xaa, 0xbb, 0xcc, 0xdd, 0xee, 0xff}",
            "mbed-mesh-api.thread-config-channel": 22,
            "mbed-mesh-api.heap-size": 14000,
            "mbed-trace.enable": false,
            "platform.stdio-convert-newlines": true,
            "platform.stdio-baud-rate": 115200
        },
        "NUCLEO_L152RE": {
            "LED": "NC",
            "BUTTON": "USER_BUTTON"
        }
    }
}
```
Top of the my main.cpp

```
#if MBED_CONF_APP_RADIO_TYPE == SPIRIT1
#include "NanostackRfPhySpirit1.h"
NanostackRfPhySpirit1 rf_phy(SPIRIT1_SPI_MOSI, SPIRIT1_SPI_MISO, SPIRIT1_SPI_SCLK,
			     SPIRIT1_DEV_IRQ, SPIRIT1_DEV_CS, SPIRIT1_DEV_SDN, SPIRIT1_BRD_LED);

#endif //MBED_CONF_APP_RADIO_TYPE
```

[image]

After debugging I can see this error on UART:
```

Build: Apr 24 2018 14:52:38
Mesh type: 3
pins not configured. Skipping the LED control.

Connecting...

++ MbedOS Fault Handler ++

FaultType: HardFault

Context:                                                  
R0   : 20014000
                                                                     R1   : 00000001
                                                                                    R2   : E000ED00
R3   : 00000000
R4   : 00000000
R5   : 00000000
R6   : 00000000
                                                                   R7   : 00000000
                                                                                  R8   : 00000000
R9   : 00000000
R10  : 00000000
R11  : 00000000
R12  : 00000000
                                                                 SP   : 20007470
                                                                                LR   : 08029BDB
PC   : 08029BDC
xPSR : 61000000
PSP  : 20007450
MSP  : 20013FD8
                                                               CPUID: 412FC230
                                                                              HFSR : 40000000
MMFSR: 00000000
BFSR : 00000000
UFSR : 00000008
DFSR : 0000000B
AFSR : 00000000
                                                                            SHCSR: 00000000
                                                                                           Mode : Thread
Priv : Privileged
Stack: PSP

Thread Info:
Current:
State: 00000002 EntryFn: 08024125 Stack Size: 00000200 Mem: 200072A0 SP: 20007460
Next:
State: 00000002 EntryFn: 08024125 Stack Size: 00000200 Mem: 200072A0 SP: 20007460
Wait Threads:
State: 00000083 EntryFn: 080274D5 Stack Size: 00000300 Mem: 20006FA0 SP: 20007238
Delay Threads:
State: 00000013 EntryFn: 08023F95 Stack Size: 00001000 Mem: 200079E0 SP: 200087F0
State: 00000053 EntryFn: 08002D1D Stack Size: 00001800 Mem: 200018B0 SP: 20003050
Idle Thread:
State: 00000002 EntryFn: 08024125 Stack Size: 00000200 Mem: 200072A0 SP: 20007460
                                     -- MbedOS Fault Handler --

```

SeppoTakalo, 2018-04-26:
@betzw Are you aware of this?

Also, do you have this HW available to test this?

Seems to be ST NUCLEO_L152RE and X-NUCLEO-IDS01A4.

drvrh, 2018-05-03:
I have ST NUCLEO_L152RE and X-NUCLEO-IDS01A4.

SeppoTakalo, 2018-05-03:
Can you attach debugger and provide backtrace from what functions caused the hard fault?

drvrh, 2018-05-04:
While debugging I found out that after calling the osDelay(milisec) function in the Thread.cpp file stops in the Assembler sentence STR R2, [R3] in line 88 in the irq_cm3.S file and I get the mentioned fault.

drvrh, 2018-05-04:
[image]

betzw, 2018-05-04:
Can you pls. post also the list of registers with their content?

drvrh, 2018-05-07:
[image]

drvrh, 2018-05-07:
I also want to know, why not only the 6lowpan stack is compiled but also the Thread and some Lora source?

[image]

[image]

SeppoTakalo, 2018-05-07:
Because that is how Mbed build system works.
Everything is compiled in. It is a job of the linker to drop the unused references.

TuomoHautamaki, 2018-05-29:
Hi @drvrh , would you be able to provide those further logs/output requested by @SeppoTakalo and @betzw

SeppoTakalo, 2018-06-04:
Closing as there is not enough information provided.

Please reopen with requested information, if the problem still persist.

closed 2018-06-04 by SeppoTakalo