ARMmbed/mbed-os-example-mesh-minimal #188: Commissioning fails to add device to Thread network

bentcooke, 2018-06-01:
Using the 5.9 oob branch, I am unable to get a mesh minimal device (TR or SED) to work with the Thread Commissioning Android app.  I have verified that the PSKd and MAC are correct for the QR code.  The commissioning app starts the process and I can see this in the logs for both the ED and BR.  The ED log shows the following before the channel scan starts over:
[DBG ][tdis]: Discover channel 26
[DBG ][mlme]: RESET MAC PIB
[DBG ][thbs]: joiner bloom : 00:00:00:08:00:00:00:00:00:00:40:00:00:00:00
[DBG ][thbs]: steering bloom : 00:00:00:00:00:00:00:00:00:00:00:00:88:00:00
[DBG ][thbs]: NO network available for scan
[DBG ][thbs]: Continue network scan

I verified the end device joins the network and works properly when using the static link config (ie no commissioning).

Attached are the logs for both the ED and the BR.
[mesh-br-K64F_commish_fail.txt](https://github.com/ARMmbed/mbed-os-example-mesh-minimal/files/2063958/mesh-br-K64F_commish_fail.txt)
[mesh-thrd-router-K64F_commish_fail.txt](https://github.com/ARMmbed/mbed-os-example-mesh-minimal/files/2063959/mesh-thrd-router-K64F_commish_fail.txt)

JuhPuur, 2018-06-04:
You're using the wrong address. The Bloom Filters is generated using the EUI-64, and if the bloom filters don't match then it's because the addresses don't match.

There appears to have been an assumption that the user knows the EUI-64 value beforehand, since the application does not print it out by default. What it does print out is the misleading mac64, which is given a default value during startup that is the same for all devices.

mesh.get_mac_address() does not return the EUI-64 until the device is succesfully connected.  This can be accomplished using static configuration. Then that value can be used for commissioning with dynamic configuration.
See [https://docs.mbed.com/docs/arm-ipv66lowpan-stack/en/latest/thread_commissioning/](https://docs.mbed.com/docs/arm-ipv66lowpan-stack/en/latest/thread_commissioning/)

> To print the MAC address, you need to connect the node to the Thread network with static configuration enabled, unless you have your own configuration for the MAC address. For example, in the mesh-minimal application, place this  printf("MAC address = %s\n", mesh.get_mac_address()); after printf("connected. IP = %s\n", mesh.get_ip_address());

This is a very laborious way to do this. The value is present in the mesh interface also with dynamic configuration before connecting and I'm looking for a way to print it out.

ashok-rao, 2018-06-04:
@bentcooke ..could you please re-check?

bentcooke, 2018-06-04:
@JuhPuur I did print the MAC addr after connection using the static link config as per the docs and had used it for generating the QR code originally.  However, the problem I am just now noticing with my setup (K64 + Atmel shield) is that the MAC (EUI-64) is changing every time the device is reset (random assignment?). 
I have tried a few different combinations of shields and K64's now with the same result.  The MAC is changing with every device reset.

JuhPuur, 2018-06-05:
Nanostack border router received updates two hours ago. Please retry with the mbed-os-5-9-0-oob branch for it.

JuhPuur, 2018-06-06:
Confirmed the issue. There is a problem with where get_mac_address() retrieves the address from. It's certainly not the RF driver.

JuhPuur, 2018-06-07:
Fix to the issue made in https://github.com/ARMmbed/mbed-os/pull/7158/

bentcooke, 2018-06-07:
@JuhPuur Thanks for the fix.  This works to print the EUID needed for commissioning.  Commissioning is now successful when the correct EUID value is used.

adbridge, 2018-06-07:
@JuhPuur The fix provided to mbed-os will not make it until 5.9.1. Please add a known issue to this example readme for 5.9.0.

closed 2018-07-12 by bentcooke