ARMmbed/mbed-os-example-mesh-minimal #190: Failing to save Thread commissioning config in NVM

bentcooke, 2018-06-07:
AFter successful commissioning, NVM does not appear to be storing the config correctly when using MESH_SD_CARD for the storage method.  When the device is reset the following message appears:

[ERR ][TjoA]: Failed to read link configuration settings from NVM

How does the SD need to be prepared for the config to be stored/read correctly?

artokin, 2018-06-08:
The failure occurs Nanostack initialisation order has changed a bit. The application tries to set file system root-path to Nanostack but because Nanostack is not yet initialised the root-path set fails. Therefore file system operation will fail. To fix this we need to modify Nanostack initialisation order.

bentcooke, 2018-06-08:
@artokin Thanks for the info on the failure mode.  I applied the fixes you mentioned and can confirm the commissioning config is being saved / restored correctly now.

ashok-rao, 2018-06-11:
Looks like issue is resolved. @bentcooke please re-open if not. Thanks all, closing this issue now.

closed 2018-06-11 by ashok-rao

ashok-rao, 2018-06-11:
Thanks @artokin, your fix PR has already been marked for release in 5.9.1. I think it's safe to close this for 5.9.0 OoB now.