ARMmbed/mbed-os-example-mesh-minimal #199: Cannot print EUI as stated in know issues in README.md

ubx-caloca, 2018-06-29:
I've imported the mbed-os-example-mesh-minimal (June 21) in the mbed compiler, then  changed the mesh-type to MESH_THREAD in the mbed_app.json.

Then I added these lines to print the eui value:
uint8_t eui64[8] = {0,0,0,0,0,0,0,0};
mesh.device_eui64_get(eui64);
printf("EUI64 address = %02x:%02x:%02x:%02x:%02x:%02x:%02x:%02x\n", eui64[0], eui64[1], eui64[2], eui64[3], eui64[4], eui64[5], eui64[6], eui64[7]);

However, when compiling the program I'm getting the following error:
Error: Class "ThreadInterface" has no member "device_eui64_get" in "main.cpp", Line: 100, Col: 10

Do I need to add or change something else? I want to print the EUI.

artokin, 2018-07-02:
I'm sorry about the inaccurate documentation in the application.

The API for `device_eui64_get` was made in PR https://github.com/ARMmbed/mbed-os/pull/7158 and it was supposed to be released in Mbed OS 5.9.1 but unfortunately the fix was delayed to Mbed OS 5.9.2. The fix is already available in the Mbed OS master branch. 

Would you please try to cherry-pick the PR to your mbed-os repo or use the master branch?

ubx-caloca, 2018-07-02:
I tried importing the application again from scratch,  the last commit that it has is the "Updating mbed-os to mbed-os-5.9.2 ... " commit. It still complaining when building that the ThreadInterface has no member device_eui64_get.

artokin, 2018-07-03:
`device_eui64_get ` is defined only for `ThreadInterface` . Therefore you need to use Thread configuration when building the application: `mbed compile -t GCC_ARM -m K64F --app-config ./configs/mesh_thread.json`.

ubx-caloca, 2018-07-03:
I'm compiling using the online Mbed IDE (for a K64F + MCR20). I'm attaching the mbed_app.json file that I used which is based on /config/mesh_thread.json.  It still does not recognize the device_eui64_get method.

[mbed_app.json.log](https://github.com/ARMmbed/mbed-os-example-mesh-minimal/files/2160633/mbed_app.json.log)

artokin, 2018-07-04:
Mbed OS 5.9.2 is not yet released and therefore online IDE is using Mbed OS 5.9.1 that does not have the required PR. Mbed OS 5.9.2 will be released soon and it will solve the problem with online IDE. 

Another option would be to try with your personal workarea. Then you can be sure that correct version of Mbed OS is used.

ubx-caloca, 2018-07-05:
Mbed OS 5.9.2 just released today. I tried with the online IDE again, imported the mbed-os-example-mesh-minimal project again, modify the mbed_app.json to be like /config/mesh_thread.json. When I compile it shows the following error:
[image]

Any suggestions?

kjbracey-arm, 2018-07-05:
That issue also reported here: https://github.com/ARMmbed/mbed-os/issues/7422

artokin, 2018-07-05:
https://github.com/ARMmbed/mbed-os/issues/7422#issuecomment-402716689 contains workaround for the problem: add macro `_RTE_` to the online IDE menu "Compile->Compile Macros" and recompile. For me it generated binary that prints correct EUI64.

I'm really sorry about this hassle, hopefully you can get the EUI64 now.

ubx-caloca, 2018-07-06:
Thanks for all your help Arto, I successfully printed the EUI value. 

I tried then to follow the instructions of https://os.mbed.com/docs/latest/reference/mesh-tech.html#thread-commissioning, there it says that you need to first create and run the firmware image with the thread-use-static-link-config=false in the mbed_app.json and add the code to print the EUI after the "printf("connected. IP = %s\n", mesh.get_ip_address());" line. 

I did that and the FRD K64F+ MCR20 printed the EUI, then compile the same code but with thread-use-static-link-config=false and setting mbed-mesh-api.thread-pskd to some value ("J01NU5"). Tried commisioning the node with the oficial commissioning app by typing the EUI that I got and the pskd that I set, but its failing seems like those values are incorrect.

I return to the image with thread-use-static-link-config=true to double check the printed EUI and I noticed that it was different.  I reset the board and again the printed EUI is different, every time I reset the node it prints a slightly different EUI value.

I also tried to move the lines to print the EUI before the mesh.connect() so that it may print the EUI in the 
thread-use-static-link-config=false image, but I'm getting a runtime error when calling the get_eui method.

I reset the board 10 times and these are the EUI that I obtained:
00:02:fc:5d:51:00:f0:09
00:02:fc:5c:51:00:f0:09
00:02:fc:5b:51:00:f0:09
00:02:fc:5f:51:00:f0:09
00:02:fc:5b:51:00:f0:09
00:02:fc:4c:51:00:f0:09
00:02:fc:5d:51:00:f0:09
00:02:fc:58:51:00:f0:09
00:02:fc:58:51:00:f0:09
00:02:fc:58:51:00:f0:09
[image]

Is there any way to avoid generating a different EUI every time the node resets, or set your own? I was under the assumption that the EUI is a unique value of the hardware and it's always the same.

artokin, 2018-07-06:
EUI64 should be constant. I tested with Atmel RF by restarting device more than 20 times: EUI64 is always constant. 

mbed-mesh-api contains method `device_eui64_set` in https://github.com/ARMmbed/mbed-os/blob/mbed-os-5.9.2/features/nanostack/mbed-mesh-api/mbed-mesh-api/ThreadInterface.h that can be used to set EUI64. However, this method is suffering about the runtime error you already noticed. The runtime error will be handled by issue: https://github.com/ARMmbed/mbed-os/issues/7391

I also tested with K64F+ MCR20 and see exactly the same problem as you did, EUI64 is not constant.

kjbracey-arm, 2018-08-20:
Please check https://github.com/ARMmbed/mcr20a-rf-driver/pull/20 for a proposed fix.

ubx-caloca, 2018-08-28:
Hello @artokin. Sure I will try it out today and put my result here. Thanks for keeping an eye on the issue.

ubx-caloca, 2018-08-28:
@artokin I imported the mbed-os-example-mesh-minimal project from scratch using the mbed online compiler, then modified the config file to create a network and print the EUI after the print the ip line, then I reseted the board multiples times to check the printed EUI value.  

I can happily say that the issue is fixed, however, I had to manually update the mcr20a-rf-driver library in the mbed online compiler to get the fix, the issue was still there in the version of the library that the mbed-os-example-mesh-minimal project gets when importing it.

ubx-caloca, 2018-08-28:
@artokin Just tried commissioning the MCR20 node using the app and to a nanostack-border-router. It working great!!.

artokin, 2018-08-29:
Thank you @ubx-caloca .  For some reason I missed the MCR20 driver update in my previous commit but it is fixed now in the https://github.com/ARMmbed/mbed-os-example-mesh-minimal/pull/211.

Closing this issue. All the fixes are available in the Mbed OS master branch and later on they will be available in the Mbed OS 5.10.

closed 2018-08-29 by artokin