ARMmbed/mbed-os-example-mesh-minimal #205: Question: Is it possible to setup a mesh network with only two router nodes?

jagomo, 2018-07-19:
I've been willing to play around with some NUCLEO_F401RE boards using the AT86RF233 shield. And I was wondering if using this example is it possible to create a mesh without the reference border router being involved (keep in mind that so far I'm just looking to communicate the boards with one another).
Reading the readme.md I found a reference to this scenario.

> "Note, that the Thread devices can form a network without the existence of the border router."

But so far, building the example setting both nodes with the following mdeb_app.json yielded no result.

```
{
    "config": {
        "radio-type":{
            "help": "options are ATMEL, MCR20, NCS36510, KW24D",
            "value": "ATMEL"
        },
        "mesh-type":{
            "help": "options are MESH_LOWPAN, MESH_THREAD",
            "value": "MESH_THREAD"
        },
            "storage-device":{
            "help": "Values for storage (Thread) devices are: MESH_SD_CARD and MESH_HEAP",
            "value": "MESH_HEAP"
        },
        "enable-led-control-example": true,
        "LED": "NC",
        "BUTTON": "NC",
        "BUTTON_MODE": "PullUp"
    },
    "target_overrides": {
        "*": {
            "target.features_add": ["NANOSTACK", "COMMON_PAL"],
            "nanostack.configuration": "thread_router",
            "mbed-trace.enable": false,
            "mbed-mesh-api.heap-size": 30000,
            "mbed-mesh-api.thread-device-type": "MESH_DEVICE_TYPE_THREAD_ROUTER",
            "platform.stdio-convert-newlines": true,
            "platform.stdio-baud-rate": 115200
        },
        "K64F": {
            "LED": "LED_RED",
            "BUTTON": "SW2"
        },
        "K66F": {
            "LED": "LED_RED",
            "BUTTON": "SW2"
        },
        "NUCLEO_F401RE": {
            "LED": "NC",
            "BUTTON": "USER_BUTTON",
            "target.stdio_uart_tx": "PA_9",
            "target.stdio_uart_rx": "PA_10",
            "platform.stdio-baud-rate": 115200,
            "platform.default-serial-baud-rate": 115200
        },
        "NUCLEO_F411RE": {
            "LED": "NC",
            "BUTTON": "USER_BUTTON"
        },
        "NCS36510": {
            "LED": "LED1",
            "BUTTON": "SW2"
        },
        "KW24D": {
            "LED": "LED1",
            "BUTTON": "SW1"
        },
        "NUCLEO_F429ZI":  {
            "LED": "LED_RED",
            "BUTTON": "USER_BUTTON",
            "BUTTON_MODE": "PullDown"
        }
    }
}
```

I just get the following output from both boards

```
Start mesh-minimal application
Build: Jul 19 2018 16:24:04
Mesh type: 4
pins not configured. Skipping the LED control.

Connecting...
```
Is there any way to configure the scenario I'm looking for? am I overseeing or doing something wrong with the configuration?

Thanks in advance.

jagomo, 2018-07-20:
Just for completeness. I also enabled embed trace and the resulting log is as follows

```
Connecting...
Start mesh-minimal application
Build: Jul 20 2018 15:39:06
Mesh type: 4
pins not configured. Skipping the LED control.

Connecting...
[DBG ][core]: Allocate Root Tasklet
[DBG ][6lo ]: P.Init

[DBG ][core]: NS Root task Init
[DBG ][sck ]: Socket Tasklet Generated
[DBG ][nsth]: connect()
[DBG ][mPKe]: Set new Key main Class
[DBG ][thbs]: Allocate Thread Tasklet
[DBG ][mMIB]: Set key 3
[DBG ][thrm]: Register Discovery device descriptor
[DBG ][mMIB]: Remove user 31 from key
[DBG ][mMIB]: Over write 31, mac16 ffff mac64: ff:ff:ff:ff:ff:ff:ff:ff, 4294967295
[DBG ][mMIB]: Set 31, mac16 ffff mac64: 35:06:fe:b8:23:d4:87:12, 0
[DBG ][m6Thread]: channel page: 0
[DBG ][m6Thread]: channel mask: 0x07fff800
[DBG ][m6Thread]: PANID 700
[DBG ][m6Thread]: channel: 22
[DBG ][m6Thread]: Mesh prefix: fd:00:0d:b8:00:00:00:00
[DBG ][mle ]: MLE service init size 31
[DBG ][tBRa]: thread_border_router_init if=1
[DBG ][ThSA]: service init interface 1, port 61631, options 0
[DBG ][ThSA]: service tasklet init
[DBG ][sck ]: Socket id 0 allocated
[DBG ][ThSA]: Service 1, Uri registration uri: a/nd
[DBG ][tBBR]: thread_border_router_init if=1
[DBG ][ThSA]: service init interface 1, port 61631, options 0
[DBG ][ThSA]: service init interface 1, port 49191, options 130
[ERR ][ThCH]: Secure CoAP unavailable - SSL library not configured, possibly due to lack of entropy source
[WARN][tBBR]: Thread border router br-service init failed
[WARN][thrm]: Thread management server init fail
[DBG ][thrd]: Interface not yet fully configured
[ERR ][m6Thread]: Bootstrap start failed, -5
[DBG ][thbs]: Thread task Init
[DBG ][mle ]: MLE Tasklet Generated
[DBG ][ThSA]: service tasklet initialised

```

SeppoTakalo, 2018-07-23:
Unfortunately Thread uses security services from Mbed TLS library, which is pretty much disabled on F401RE due to the lack of support for entropy source.

Most important line in your last log:
```
[ERR ][ThCH]: Secure CoAP unavailable - SSL library not configured, possibly due to lack of entropy source
```

Other than that, it should be possible to create stand-alone Thread network, as the comissioning is disabled by default and all devices share the same master key by default.
Then you could use link local or mesh-local prefixes `fd00:db8::XXXX` to talk with devices.

If you have any other devices that has TRNG support, this should work.

SeppoTakalo, 2018-07-25:
@jagomo Thanks for your information.

I'm tempted to reject the PR if we would enable NULL_ENTROPY. It is way too easy to break good security by giving examples of bad usage.
For testing purposes, it is good to know, but user should be aware of what they are doing when enabling it. Therefore I'm not going to give out just copy&paste macros. Those who know what they are doing, find the information from its original sources, as you did.

For F401RE, I'm happy to accept PR if you want those pin and led names to be applied.

I'll close this issue for now, as the original issue is now solved. Thank you.

closed 2018-07-25 by SeppoTakalo