ARMmbed/mbed-os-example-mesh-minimal #217: Support for KW41Z

vervaekejonathan, 2018-09-14:
Is KW41Z supported? If not, will it be supported and when?
Thanks!

TuomoHautamaki, 2018-09-17:
Hi @vervaekejonathan ,

Support has been coming but not quite there yet. We need to re-open this one [6622](https://github.com/ARMmbed/mbed-os/pull/6622)
once the needed test enablers land to mbed OS (target is mbed OS 5.1.x). PR already open [7996](https://github.com/ARMmbed/mbed-os/pull/7996)

Regards,
Tuomo

vervaekejonathan, 2018-09-17:
Thanks for bumping the finalization!
Will there be any documentation available to get started? Or is it as simple as setting the right target during building?

JuhPuur, 2018-09-18:
See the [documentation](https://github.com/ARMmbed/mbed-os/blob/fe4653309db25191bb2a64b0f342ad3a01b1cde8/TEST_APPS/device/nanostack_mac_tester/README.md), section "Automated mode".

JuhPuur, 2018-09-19:
One important thing to remember(and probably a big omission in the instructions) is that the target needs to be added to tools/test_configs/targets.json before mbed-cli will accept the test config for that target and begin compiling. In the case of KW41Z, adding the following to the list is needed:

```
    "KW41Z": {
        "default_test_configuration": "NONE",
        "test_configurations": ["NANOSTACK_MAC_TESTER"]
    }
```

SeppoTakalo, 2018-12-05:
Closing, as this application is not platform specific.

When Mbed OS supports KW41Z, then the mesh application will support as well.

closed 2018-12-05 by SeppoTakalo