ARMmbed/mbed-os-example-mesh-minimal #250: Support Thunderboard Sense 2

andynnd, 2019-04-05:
Hi. Can anyone run this example in Thunderboard Sense 2 successful? I built and flashed it but the serial terminal didn't show anything. I find that mbed support the Silabs RF driver already as below link.
https://github.com/ARMmbed/mbed-os/tree/master/targets/TARGET_Silicon_Labs/TARGET_SL_RAIL/efr32-rf-driver
Thanks.

andynnd, 2019-04-05:
Thanks, ciarmcom. Maybe I cannot access that. Can I know when it will be available for developer?

0xc0170, 2019-04-05:
> Thanks, ciarmcom. Maybe I cannot access that. Can I know when it will be available for developer?

@andynnd it's internal jira reference created by a bot. It should probably have a suffix -bot to be more descriptive.

SeppoTakalo, 2019-04-08:
I tried the same with GCC and serial terminal is empty.
Then I tried with Mbed Studio, which used Arm Compiler 6, and the serial terminal shows HardFault.

So clearly something wrong.

SeppoTakalo, 2019-04-08:
ah..
```
++ MbedOS Error Info ++
Error Status: 0x80FF0144 Code: 324 Module: 255
Error Message: Assertion failed: pin != NC
Location: 0x29F49
File: ./mbed-os/targets/TARGET_Silicon_Labs/TARGET_EFM32/gpio_api.c+66
Error Value: 0x0                                                                                          
Current Thread: main  Id: 0x2000D5EC Entry: 0x2A48D StackSize: 0x1000 StackMem: 0x200025E8 SP: 0x20003524 
For more info, visit: https://mbed.com/s/error?error=0x80FF0144&tgt=TB_SENSE_12                           
-- MbedOS Error Info --                                                                                   
 ```

I'll check the pin configuration.

SeppoTakalo, 2019-04-08:
Following *mbed_app.json* seems to work:
```
{
    "config": {
        "enable-led-control-example": true,
        "LED": "NC",
        "BUTTON": "NC",
        "RELAY_CONTROL": "NC",
        "BUTTON_MODE": "PullUp"
    },
    "target_overrides": {
        "*": {
            "nanostack.configuration": "lowpan_router",
            "nsapi.default-mesh-type": "LOWPAN",
            "platform.stdio-convert-newlines": true,
            "platform.stdio-baud-rate": 115200
        },
        "TB_SENSE_12": {
            "LED": "LED1",
            "BUTTON": "BTN1",
            "RELAY_CONTROL": "PA8"
        }
    }
}
```

Results:
```
Start mesh-minimal application
Build: Apr  8 2019 13:22:05
Mesh type: MBED_CONF_NSAPI_DEFAULT_MESH_TYPE
Mbed OS version: 5.12.0
Connecting...
Connected. IP = 2001:999:71:433a:fd57:b00:b258:befe

 ```

andynnd, 2019-04-08:
Thanks, @SeppoTakalo. I set the same values with yours but it didn't work last time. However, it works now.

closed 2019-04-08 by andynnd