ARMmbed/mbed-os-example-mesh-minimal #258: 5.13 OOB branch has old Mbed OS 5.13.0-rc1 tagged in mbed-os.lib

korjaa, 2019-06-06:
[image]

korjaa, 2019-06-06:
https://github.com/ARMmbed/mbed-os-example-mesh-minimal/pull/259

SeppoTakalo, 2019-06-06:
I don't understand.. the 5.13.0-rc2 is submitted here.

https://github.com/ARMmbed/mbed-os-example-mesh-minimal/commit/b7d8bebfbe6f9cae5dc054f66659d02940e2c0c9

Also, your screenshot does not make sense.. on top, there is 5.13.0-rc2 tag. Why the rc-1 is circled?

korjaa, 2019-06-07:
The circled version was tagged on the  mbed-os-5.13_oob branch which I was instructed to test.

b7d8beb is merged to mbed-os-5.13

michalpasztamobica, 2019-08-28:
The branch has just been updated, rendering this issue out-of-date: https://github.com/ARMmbed/mbed-os-example-mesh-minimal/commit/2b82355acd4fd9510b839af0ae4e3a32f7e57733

closed 2019-08-28 by michalpasztamobica