ARMmbed/mbed-os-example-mesh-minimal #264: 5.13 OOB - Atmel AT86RF233 firefly link is broken.

korjaa, 2019-06-07:
https://firefly-iot.com/product/firefly-arduino-shield-2-4ghz/ results in "Unable to connect"

SeppoTakalo, 2019-06-07:
Seems that L-Tek has changed their URLs.

This one is working:
https://l-tek.si/web-shop/l-tek-6lowpan-arduino-shield-2-4ghz/

@artokin will you update?

michalpasztamobica, 2019-06-18:
l-tek's 6lowpan arduino shield mentioned above uses AT86RF212, not AT86RF233. 
AT86RF212 is a sub-GHz band device, while AT86RF233 is 2.4 GHz.

michalpasztamobica, 2019-06-18:
Confirmed with @kimlep01 from Infrastructure team, that this is the right link, indeed.

michalpasztamobica, 2019-06-19:
The confusion has been clarified with the web shop, they updated the schematic and now it clearly shows that AT86RF233 has been used.

michalpasztamobica, 2019-08-28:
Fixed with https://github.com/ARMmbed/mbed-os-example-mesh-minimal/pull/269

closed 2019-08-28 by michalpasztamobica

nbrunner, 2022-01-27:
The link https://www.l-tek.com/web-shop/l-tek-6lowpan-arduino-shield-2-4ghz/ return to a "Oops! That page can’t be found."