ARMmbed/mbed-os-example-mesh-minimal #280: How to keep the mbed-os version of mbed-os-example-mesh-minmal up to date

JayJaehoonRhee, 2019-09-03:
Recently I've cloned mbed-os-example-mesh-minial from github using "mbed import" command. As I know, the latest version of mbed-os is released as 5.13.4.  But when I compared mbed-os from https://github.com/ARMmbed/mbed-os with mbed-os from https://github.com/ARMmbed/mbed-os-example-mesh-minimal, source tree Is very different. I think it might be not same version of these mbed-os sources.
My previous efforts to keep the mbed-os version of mbed-os-example-mesh-minial up-to-date are below.

cd mbed-os
git fetch
git tag | grep "5.13"
git checkout mbed-os-5.13.4
git checkout -b my_branch
git log

=====================================================================
my git log
commit 1bf6b20df9d3cd5f29f001ffc6f0d0fcbbb96118
Merge: 808c450 75ca176
Author: Martin Kojtal <martin.kojtal@arm.com>
Date:   Mon Aug 26 10:03:59 2019 +0200

    Merge pull request #11308 from ARMmbed/release-candidate
    
    Release candidate for mbed-os-5.13.4

commit 75ca176183a9002150247bdc184855248d3869e5
Author: adbridge <anna.bridge@arm.com>
Date:   Fri Aug 23 14:33:43 2019 +0100

    Revert "Add Unittest equeue tests"
    
    This reverts commit 7ea8e0e89a155f972d08c7afa3fd3e288906e07a.

commit 42db954d06fa1930786fb5dc91b569ae5394edf8
Author: Reda Maher <redamaher@riotmicro.com>
Date:   Mon Jul 29 21:17:15 2019 +0200

    Add Riot Micro cellular module targets

commit d81b1cfa9777de7a1ff327bebc0e297eb0697940
Author: Reda Maher <redamaher@riotmicro.com>
Date:   Mon Jul 29 21:15:02 2019 +0200

    Cellular: Support Riot Micro cellular device

commit 5fa34b387a3c5bd837348a596a5309af2b2aa144
Author: adbridge <anna.bridge@arm.com>
Date:   Fri Aug 23 11:41:11 2019 +0100

    Update Mbed version block

commit fddc2b1dd54f16a23c3b19ba5c1eaa29a4f97b31
Author: Mahesh Mahadevan <mahesh.mahadevan@nxp.com>
Date:   Thu Aug 22 10:47:43 2019 -0500

    MCUXpresso: Update EDMA drivers to fix warnings
    
    Signed-off-by: Mahesh Mahadevan <mahesh.mahadevan@nxp.com>

commit 1d9b067212678824030703df04b89370533b6a19
Author: Mahesh Mahadevan <mahesh.mahadevan@nxp.com>
Date:   Thu Aug 22 08:26:11 2019 -0500

    MCUXpresso: Update the SDK LPUART drivers
    
    Use the latest driver from K66F
    
    Signed-off-by: Mahesh Mahadevan <mahesh.mahadevan@nxp.com>

commit eda75fd02a27d0d26bbed4953abb19ed778a5a8c
Author: Mahesh Mahadevan <mahesh.mahadevan@nxp.com>
Date:   Wed Aug 21 17:18:34 2019 -0500

    MCUXpresso: Update I2C SDK driver
    
    Use the K66F driver that has the latest
    
    Signed-off-by: Mahesh Mahadevan <mahesh.mahadevan@nxp.com>

commit 8022ca918fd173e5f1eb78c784b09d5cbf6a6ae4
Author: Mahesh Mahadevan <mahesh.mahadevan@nxp.com>
Date:   Wed Aug 21 15:38:37 2019 -0500
======================================================================
 
What should I do more?
If it's right, could you tell me the corresponding git command?

SeppoTakalo, 2019-09-03:
Hi @JayJaehoonRhee 
All Mbed OS example applications now follow the model where `master` branch is development branch, and might not follow Mbed OS releases.

For every feature release version, there is branch existing. So for Mbed OS 5.13 release, this application have `mbed-os-5.13` branch.

You can follow this release by doing the `mbed import` normally, then `cd <app>; mbed update mbed-os-5.13`

This will do git-checkout of branch called `mbed-os-5.13` and update all libraries toi point correct versions.

JayJaehoonRhee, 2019-09-03:
Hi @SeppoTakalo 
Even though I followed as you commented, the source tree s of mesh-minimal-example is still different from https://github.com/ARMmbed/mbed-os.
Does "mbed update" command only update the libraries?
I just want to make the source tree of mbed-os 5.13.4 released the same as the source tree of mbed-os of mesh-minimal-example.

michalpasztamobica, 2019-10-09:
Hi @JayJaehoonRhee , sorry to keep you waiting for so long.
Not sure if that's the issue, but what you actually see when you go straight to https://github.com/ARMmbed/mbed-os and view the revision history is not any particular release, but simply the most recent `master` branch, as Seppo explained.

If you want to be in line with what you see there you need to run: `mbed update master`.

michalpasztamobica, 2019-12-03:
@JayJaehoonRhee , did you manage to solve this issue or is it still open?

michalpasztamobica, 2019-12-04:
@JayJaehoonRhee I am closing this issue, but if you can provide more information or need more support, feel free to reopen.

closed 2019-12-04 by michalpasztamobica