ARMmbed/mbed-os-example-mesh-minimal #288: Wi-SUN BR and router do not connect well

andrewc-arm, 2019-09-30:
Hi,

I tried `mbed-os-5.14.0` for both nanostack-border-router and mbed-os-example-mesh-minimal but they failed to connect each other. Could you please help me on which steps I could have missed?

## Logs

### 2019-09-30 Failure logs
[teraterm_20190930.1607 - border router.log](https://github.com/ARMmbed/mbed-os-example-mesh-minimal/files/3669427/teraterm_20190930.1607.-.border.router.log)
[teraterm_20190930.1631 - router.log](https://github.com/ARMmbed/mbed-os-example-mesh-minimal/files/3669428/teraterm_20190930.1631.-.router.log)

## How to reproduce

### Border Router setup
* Use K64F + S2LP RF (HW modified for SPI CLK)
* get [nanostack-border-router](https://github.com/ARMmbed/nanostack-border-router) of `mbed-os-5.14.0`
* `cp configs/Wisun_Stm_s2lp_RF.json mbed_app.json`
* change the regional code for Korea
```
		"regulatory-domain": 9,
```
* Compile
```
mbed compile -j4 -v --profile debug -t GCC_ARM -m K64F > build_log.txt
```

### Router setup
* Use NUCLEO-F429ZI + S2LP RF (HW modified for SPI CLK)
* get [mbed-os-example-mesh-minimal](https://github.com/ARMmbed/mbed-os-example-mesh-minimal) of `mbed-os-5.14.0`
* cp configs/mesh_wisun_S2LP.json mbed_app.json
* change the trace
```
            "mbed-trace.enable": true,
```
* Compile
```
mbed compile -j4 -v --profile debug -t GCC_ARM -m NUCLEO_F429ZI > build_log.txt
```

### Test
* Connect BR to IPv6 tunneler. Make sure it can acquire public IPv6 address.
* Start router.

andrewc-arm, 2019-09-30:
Hi,

Do I need to set `"regulatory-domain": 9,` on `Router setup` as well?

andrewc-arm, 2019-09-30:
Another question. Due to json configuration, there seems to be double regulatory domains.
```
BUILD/K64F/GCC_ARM-DEBUG/mbed_config.h:47:#define MBED_CONF_APP_REGULATORY_DOMAIN                                       9                                                                                                // set by application
BUILD/K64F/GCC_ARM-DEBUG/mbed_config.h:173:#define MBED_CONF_MBED_MESH_API_WISUN_REGULATORY_DOMAIN                       3                                                                                                // set by library:mbed-mesh-api
```
Is this expected or should there be integrity on domains?

mikter, 2019-10-03:
Hi,

I looked at the logs and the mesh minimal log has not waited long enough that the network can start. Wi-SUN networks are slow and the network formation takes normally about 10-15 minutes and can take even longer if there is missed packets during startup.

Reason for this is that networks are designed to have hundreds of devices and as the individual device does not know how large the surrounding is it cannot start spamming the channels.

so could you verify that after 15 minutes the network forms or doesn't ?

andrewc-arm, 2019-10-04:
Hi, @mikter 
I let one BR and two routers to run for more than 30 minutes but they didn't connect. Here are the logs.
[teraterm_20191004.1411 - Copy.log](https://github.com/ARMmbed/mbed-os-example-mesh-minimal/files/3689132/teraterm_20191004.1411.-.Copy.log)
[teraterm_20191004.1356 - Copy.log](https://github.com/ARMmbed/mbed-os-example-mesh-minimal/files/3689133/teraterm_20191004.1356.-.Copy.log)
[teraterm_20191004.1357 - Copy.log](https://github.com/ARMmbed/mbed-os-example-mesh-minimal/files/3689134/teraterm_20191004.1357.-.Copy.log)

mikter, 2019-10-04:
Problem is that there is a different configuration in border router and in router.
You can see that from logs by comparing these lines
BR
[2019-10-04 13:56:39.460] [INFO][mlme]: RF config update:
[2019-10-04 13:56:39.468] [INFO][mlme]: Frequency(ch0): 917300000Hz
[2019-10-04 13:56:39.472] [INFO][mlme]: Channel spacing: 400000Hz
[2019-10-04 13:56:39.476] [INFO][mlme]: Datarate: 150000bps
[2019-10-04 13:56:39.483] [INFO][mlme]: Number of channels: 16
[2019-10-04 13:56:39.484] [INFO][mlme]: Modulation: 4
[2019-10-04 13:56:39.490] [INFO][mlme]: Modulation index: 0

From router
[2019-10-04 13:57:15.118] [INFO][mlme]: RF config update:
[2019-10-04 13:57:15.121] [INFO][mlme]: Frequency(ch0): 863100000Hz
[2019-10-04 13:57:15.127] [INFO][mlme]: Channel spacing: 200000Hz
[2019-10-04 13:57:15.131] [INFO][mlme]: Datarate: 150000bps
[2019-10-04 13:57:15.136] [INFO][mlme]: Number of channels: 35
[2019-10-04 13:57:15.141] [INFO][mlme]: Modulation: 4
[2019-10-04 13:57:15.144] [INFO][mlme]: Modulation index: 0

So check configurations off both devices and make them match. Also verify that you have same version of Wi-SUN stack on both devices

andrewc-arm, 2019-10-04:
Hmm... It's strange. I used `mbed-os-5.14.0` on both softwares. Anyway, I will try to match them as you commented. Thanks for the quick support!

mikter, 2019-10-04:
As I read your error description you only change the regulatory domain for Border router, and it needs to be changed on both

andrewc-arm, 2019-10-04:
Thanks for the feedback. I will investigate and get back to you soon.

andrewc-arm, 2019-10-07:
@mikter 
I verified that reverting back `"regulatory-domain": 3,` changed back the frequency to `863100000Hz` in the border router. This means router application doesn't change the frequency from setting the `"regulatory-domain": 9,`. Could you or your team please enable `"regulatory-domain": 9,` on this mesh-minimal application?
(CC: @AnttiKauppila )

andrewc-arm, 2019-10-07:
Hi, @mikter 
Thanks for the guidance. I finally got the connection using `"regulatory-domain": 3,`. 😃 
[teraterm_20191007.1812-2 - Copy.log](https://github.com/ARMmbed/mbed-os-example-mesh-minimal/files/3696464/teraterm_20191007.1812-2.-.Copy.log)
```
[2019-10-07 18:34:14.749] [DBG ][WSND]: app_parse_network_event() 0
[2019-10-07 18:34:14.754] [INFO][WSND]: Wi-SUN bootstrap ready
[2019-10-07 18:34:14.757] Connected. IP = 2001:470:f857:1fe9:94bf:b293:d974:9d5b
[2019-10-07 18:34:14.764] [DBG ][example]: start_mesh_led_control_example()
```
I am going to do just simple demo locally and do I need to set `"regulatory-domain": 9,`? Could you kindly explain what happens if I don't? 

Thanks!

mikter, 2019-10-07:
For simple demo locally You are using frequency bands that are not intended for that usage, but on low power and shielded environment it is ok. 

If you add the regulatory domain setting in both applications it should work on regulatory domain 9, but it must be on both applications

andrewc-arm, 2019-10-08:
Hi, @mikter 
I changed the issue title as `Wi-SUN regulatory-domain not implemented for router`.
Even if I change the `regulatory-domain` in the mesh-minimal Wi-SUN application, it does not change the frequency band. Border router application is fine, but mesh-minimal application is not fine.
Could you please guide on how we router Wi-SUN application (mesh-minimal) can change `regulatory-domain`? Just adding this line didn't change the setting.
```
		"regulatory-domain": 9,
```

mikter, 2019-10-08:
The correct configuration added to target overrides is

`
            "mbed-mesh-api.wisun-regulatory-domain": 9,
`

andrewc-arm, 2019-10-08:
@mikter 
Could you please review this log? There is an error and mesh-minimal/BR does not work even I set BR and R the domain 9.
[mesh_minimal_domain9_failure01.zip](https://github.com/ARMmbed/mbed-os-example-mesh-minimal/files/3701226/mesh_minimal_domain9_failure01.zip)
```
[2019-10-08 16:28:29.752] [INFO][wsps]: EAPOL target: c6:d1:8d:b4:b4:b5:54:4b
[2019-10-08 16:28:29.761] [INFO][ksep]: Initial EAPOL-Key send, PMKID not set PTKID not set GTKL 0
[2019-10-08 16:28:30.316] [ERR ][mMCp]: MAC tx fail
```

mikter, 2019-10-11:
Could you check that Border router and mesh minimal has the same versions on Nanostack and rf driver as there seem to be only one way link between the devices.

andrewc-arm, 2019-10-15:
@mikter 

I checked again and the versions are the same. When I run with the domain 3, it works well. When I only change the domain to 9 then it fails.

FYI. This is the version of the border router.
```
andcho01@bc-a1-1-3:~/code/nanostack-border-router$ git status
HEAD detached at mbed-os-5.14.0
```

This is the version of mesh minimal.
```
andcho01@bc-a1-1-3:~/code/mbed-os-example-mesh-minimal$ git status
HEAD detached at mbed-os-5.14.0
```

andrewc-arm, 2019-10-15:
@mikter 
I checkout the domain 9 and still I get the MAC TX fail.
[image]
[WiSUN_5.14.0_domain9_fail02.zip](https://github.com/ARMmbed/mbed-os-example-mesh-minimal/files/3727634/WiSUN_5.14.0_domain9_fail02.zip)

mikter, 2019-10-15:
@andrewc-arm We found the issue and it was merged in to mbed-os master The fix will be released in 5.14.1

andrewc-arm, 2019-10-16:
@mikter 
I tried with the very latest mbed-os (https://github.com/ARMmbed/mbed-os/commit/ea3ead034992f72453836e2d6f7efdf7346cde06) and I got mesh-minimal working with regulatory-domain 9!
[WiSUN_master_domain9_success01.zip](https://github.com/ARMmbed/mbed-os-example-mesh-minimal/files/3732181/WiSUN_master_domain9_success01.zip)
Thank you for your support!

closed 2019-10-16 by andrewc-arm