ARMmbed/mbed-os-example-mesh-minimal #290: How to increase distance of Wi-SUN?

andrewc-arm, 2019-10-11:
Hi,
For the Wi-SUN + PDMC demo, Korea team would like to increase the distance between devices. Could you please guide us on how we can increase the distance? Do we need to tweak HW and SW?
Thanks,

(CC: @taeily )

mikter, 2019-11-08:
Hi,

There is two ways that you can force topologies.

with HW You can hardwire antennas with attenuators and splitters.

With SW you could make direct call for nanostack filtering API and only allow packets from specific device. But this is really hard to make as some devices randomises the extended address during startup making compile time filtering impossible.

andrewc-arm, 2019-11-11:
Thanks for the feedback!

closed 2019-11-11 by andrewc-arm