ARMmbed/mbed-os-example-mesh-minimal #299: Wi-SUN TLS error

mcuxmx, 2020-03-18:
Hi,
I'm using the online compiler build NUCLEO-F411RE+S2LP as router from [mbed-os-example-mesh-minimal](https://github.com/ARMmbed/mbed-os-example-mesh-minimal) and DISCO-F769NI+S2LP as border router from [nanostack-border-router](https://github.com/ARMmbed/nanostack-border-router).
the two device connect fail when EAP-TLS handshake.

[DBG ][tlsp]: TLS: start, eui-64: a2:33:d7:1c:a4:46:99:50
[DBG ][tlsp]: TLS QUEUE add index: 0, eui-64: a2:33:d7:1c:a4:46:99:50
[ERR ][tlsp]: TLS: error, eui-64: a2:33:d7:1c:a4:46:99:50
[DBG ][tlsp]: TLS: finish, eui-64: a2:33:d7:1c:a4:46:99:50
[ERR ][eapa]: EAP-TLS: handshake failed
[DBG ][tlsp]: TLS QUEUE remove last, eui-64: a2:33:d7:1c:a4:46:99:50

I add my log and config files, please take a look, thanks!

[br.log](https://github.com/ARMmbed/mbed-os-example-mesh-minimal/files/4346619/br.log)
[router.log](https://github.com/ARMmbed/mbed-os-example-mesh-minimal/files/4346640/router.log)
[br_config.txt](https://github.com/ARMmbed/mbed-os-example-mesh-minimal/files/4346641/br_config.txt)
[node_config.txt](https://github.com/ARMmbed/mbed-os-example-mesh-minimal/files/4346644/node_config.txt)

mikaleppanen, 2020-04-23:
Could be e.g. due to memory exhaustion in the node. 

Could you enable TLS traces and capture logs for further analysis.

Traces are activated by adding flags:

#define MBEDTLS_SSL_DEBUG_ALL
#define MBEDTLS_DEBUG_C

To the border router and mesh minimal TLS configuration files:

https://github.com/ARMmbed/mbed-os-example-mesh-minimal/blob/master/mbedtls_wisun_config.h
https://github.com/ARMmbed/nanostack-border-router/blob/master/source/mbedtls_wisun_config.h

Trace printing is enabled by enabling the TLS_SEC_PROT_LIB_TLS_DEBUG define on TLS module on mbed-os for both node and border router (as default the define is in comments):

https://github.com/ARMmbed/mbed-os/blob/master/features/nanostack/sal-stack-nanostack/source/Security/protocols/tls_sec_prot/tls_sec_prot_lib.c

#define TLS_SEC_PROT_LIB_TLS_DEBUG       // Enable mbed TLS debug traces

After these changes TLS traces should show on logs.

mcuxmx, 2020-04-26:
@mikaleppanen 
Thank you very much for your reply, I have added DEBUG for TLS.
On the BR side,  the log prompt bad client hello message

[INFO][app ]: Heap size: 59952, Reserved: 23916, Reserved max: 24648, Alloc fail: 0
[INFO][brro]: Backhaul interface addresses:
[INFO][brro]:  [0] fe80::280:e1ff:fe31:1e
[INFO][brro]: RF interface addresses:
[INFO][brro]:  [0] fe80::b1:c6b8:b876:c00
[INFO][brro]:  [1] fd00:6172:6d00:0:b1:c6b8:b876:c00
[INFO][eapa]: EAP-TLS: send REQ type TLS id 2 flags 20 len 10, eui-64: a2:33:d7:1c:a4:46:99:50
[INFO][eapa]: EAP-TLS: recv RESPONSE type TLS id 2 flags 0 len 15, eui-64 a2:33:d7:1c:a4:46:99:50
[DBG ][tlsp]: TLS: start, eui-64: a2:33:d7:1c:a4:46:99:50
[DBG ][tlsp]: TLS QUEUE add index: 0, eui-64: a2:33:d7:1c:a4:46:99:50
[DBG ][tlsl]: 2 ../mbed-os/features/mbedtls/src/ssl_srv.c 4456 server state: 0

[DBG ][tlsl]: 2 ../mbed-os/features/mbedtls/src/ssl_tls.c 3828 => flush output

[DBG ][tlsl]: 2 ../mbed-os/features/mbedtls/src/ssl_tls.c 3840 <= flush output

[DBG ][tlsl]: 2 ../mbed-os/features/mbedtls/src/ssl_srv.c 4456 server state: 1

[DBG ][tlsl]: 2 ../mbed-os/features/mbedtls/src/ssl_tls.c 3828 => flush output

[DBG ][tlsl]: 2 ../mbed-os/features/mbedtls/src/ssl_tls.c 3840 <= flush output

[DBG ][tlsl]: 2 ../mbed-os/features/mbedtls/src/ssl_srv.c 1323 => parse client hello

[DBG ][tlsl]: 2 ../mbed-os/features/mbedtls/src/ssl_tls.c 3609 => fetch input

[DBG ][tlsl]: 2 ../mbed-os/features/mbedtls/src/ssl_tls.c 3770 in_left: 0, nb_want: 5

[DBG ][tlsl]: 2 ../mbed-os/features/mbedtls/src/ssl_tls.c 3794 in_left: 0, nb_want: 5

[DBG ][tlsl]: 2 ../mbed-os/features/mbedtls/src/ssl_tls.c 3795 ssl->f_recv(_timeout)() returned 5 (-0xfffffffb)

[DBG ][tlsl]: 2 ../mbed-os/features/mbedtls/src/ssl_tls.c 3815 <= fetch input

[DBG ][tlsl]: 4 ../mbed-os/features/mbedtls/src/ssl_srv.c 1355 dumping 'record header' (5 bytes)

[DBG ][tlsl]: 4 ../mbed-os/features/mbedtls/src/ssl_srv.c 1355 0000:  41 6e 6f 6e 79                                   Anony

[DBG ][tlsl]: 3 ../mbed-os/features/mbedtls/src/ssl_srv.c 1367 client hello v3, message type: 65

[DBG ][tlsl]: 1 ../mbed-os/features/mbedtls/src/ssl_srv.c 1371 bad client hello message

[ERR ][tlsp]: TLS: error, eui-64: a2:33:d7:1c:a4:46:99:50
[DBG ][tlsp]: TLS: finish, eui-64: a2:33:d7:1c:a4:46:99:50
[ERR ][eapa]: EAP-TLS: handshake failed
[DBG ][tlsp]: TLS QUEUE remove last, eui-64: a2:33:d7:1c:a4:46:99:50
[INFO][wsbs]: Send PAN advertisement

mikaleppanen, 2020-04-27:
Yes, the client hello sent by the node is too small to be valid:

[INFO][eapa]: EAP-TLS: recv RESPONSE type TLS id 2 flags 0 len 15, eui-64 a2:33:d7:1c:a4:46:99:50

Length is 15, it should be over 100. 

Do you have the node TLS log?

mcuxmx, 2020-04-27:
I use the online compiler to compile the project, I don't know how to modify the macro of TSL DEBUG you mentioned earlier.
So I export the project as vscode-gcc-arm project. And modified the TLS DEBUG macro, the project has not been recompiled. I found that "mbedtls_wisun_config.h" is not included in mbed_config.h, I tried to add a macro
`#define MBEDTLS_USER_CONFIG_FILE "mbedtls_wisun_config.h"`
The project was recompiled, but there is still no output of TLS debugging information

bearxiong99, 2020-08-01:
do you fixed the problem , i met the same problem

bearxiong99, 2020-08-01:
> I use the online compiler to compile the project, I don't know how to modify the macro of TSL DEBUG you mentioned earlier.
> So I export the project as vscode-gcc-arm project. And modified the TLS [...]

do you fixed this problem ?

bearxiong99, 2020-08-01:
> TLS traces should show on logs.
@mikaleppanen 
does the  nanostack-border-router need to connect with Pelion  during the EAP-TLS ?