ARMmbed/mbed-os-example-mesh-minimal #330: Update LED and BUTTON pin names according to standard

jeromecoutant, 2021-04-06:
### Description of defect

Please update mbed_app.json according to latest pin standard.

- Avoid LED_RED => use LED1
- Use BUTTON1

@MarceloSalazar @gpsimenos 

Thx

#### Target(s) affected by this defect ?

ALL

#### Toolchain(s) (name and version) displaying this defect ?

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os master

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

#### How is this defect reproduced ?

MarceloSalazar, 2021-04-12:
Thanks @jeromecoutant , we'll look into this in the upcoming days

gpsimenos, 2021-05-04:
Addressed in #334

jeromecoutant, 2021-05-07:
> Addressed in #334

Github tip :-)
In #334 description, you should have added line "fix #330",
so Github could close automatically linked issue when merging :-)

closed 2021-05-07 by jeromecoutant